jerryscript-project/jerryscript #1916: Issue in 03.API-EXAMPLE.md

Osamu-Nakamura, 2017-07-13:
It seems that there is the issue in the following example function stated in https://github.com/jerryscript-project/jerryscript/blob/master/docs/03.API-EXAMPLE.md#step-5-description-of-jerryscript-value-descriptors.

```c
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include "jerryscript.h"

static void
print_value ( const jerry_value_t value)
{
(snip)
  /* String value */
  else if (jerry_value_is_string (value))
  {
    /* Determining required buffer size */
    jerry_size_t req_sz = jerry_get_string_size (value);
    jerry_char_t str_buf_p[req_sz];

    jerry_string_to_char_buffer (value, str_buf_p, req_sz);
    str_buf_p[req_sz] = '\0';
(snip)
```
Should the size of str_bug_p be not **req_sz** but **req_sz+1**? Otherwise, the stack area is thought to become corrupted.

LaszloLango, 2017-07-13:
Yes it should, please open a PR with the fix.

Osamu-Nakamura, 2017-07-13:
I issued #1919 for the fix.

bzsolt, 2017-07-17:
Merged in https://github.com/jerryscript-project/jerryscript/commit/bb8ac29cce49ad6f2adb2af6c9558bdef6def515

closed 2017-07-17 by bzsolt