jerryscript-project/jerryscript #1929: check JScode error on JerryScript

k-harada413, 2017-07-19:
Hi,

I have a question.
Please tell me where is problem on JScode via Jerryscript.
I try to build JScode on JerryScript.
But this error(?) occurs and JScode doesn't run.

> start
> ret1 = 0x20003c07
> ret2 = 0x20003c2f

I think `ret2` is byte code address. But it doesn't teach me where is problem on JScode.
(Its JScode can run my environment. Only JScode. not use JerryScript)

I want to know where is problem of my JScode on JerryScript.

Thank you for being patient with my English.

LaszloLango, 2017-07-19:
Hi @k-harada413,

what is JScode? Could you tell us more info about what you are trying to do?

k-harada413, 2017-07-20:
I'm sorry I didn't explain it enough.  
"JScode" is JavaScript.

I tried that:
First,  I assigned a Javascript code that is so long on `jerry_char_t`. 　
Next, `jerry_parse` and `jerry_run`.
The code stopped half way of `jerry_run` and the message is denoted.

```
start
ret1 = 0x20003c07
ret2 = 0x20003c2f
```

I want to know where is cause of stopping code in `jerry_char_t`.
It's so difficult to specify point of cause from `0x20003c2f`.

zherczeg, 2017-07-20:
Could you share your code and your main function?

> I want to know where is cause of stopping code in `jerry_char_t`.

I don't really understand this sentence.

akosthekiss, 2017-07-20:
@k-harada413 I've never seen such output from JerryScript itself. Are you sure the above output is emitted by the engine? To be able to help, we should be able to reproduce your issue. Therefore, please, describe all relevant information needed for reproduction:
- What is the build and the execution environment you are working with? Arch: X86/ARM/..., OS: Linux/OSX/..., compiler: gcc/clang/... (Arch and OS for build and exec environment, if they are different)
- What is the version of JerryScript you are using? Git hash is fine
- Are you using the command line tool `jerry` (source in jerry-main) to execute JS? If not, did you try to reproduce the issue there?
- What is the minimal JS program that still triggers the issue? (Please, try minimization by hand at least, or use some automated tools, perhaps)
- Have you tried enabling error messages during the build of JerryScript? (Although, the reported error output does not seem like it would help.)

k-harada413, 2017-07-20:
Sorry, I can't show the code using now.
I want to know how to specify which instruction causes error in JavaScript code.

Example, I want to know cause of error below codes.
(I know that not `prin` but `print`.)

```cpp
int main() {
    mbed::Serial pc(USBTX, USBRX);
    pc.baud(115200);

    JERRY_USE_MBED_LIBRARY(base);

    jsmbed_js_init();

    printf("start\r\n");

    const jerry_char_t script[] = "var hl = \"hello\";var wo = \"world\";prin(hl + \" \" + wo);";
    size_t script_size = strlen ((const char *) script);

    jerry_value_t parse_ret_val = jerry_parse (script, script_size, false);
    jerry_value_t returned_value = jerry_run(parse_ret_val);

    printf("ret1 = 0x%x\r\n",parse_ret_val);
    printf("ret2 = 0x%x\r\n",returned_value);

    mbed::js::event_loop();

    return 0;
}
```
(My code using now has a very long string of JavaScript.)

then console log is
```
start
ret1 = 0x20003beb
ret2 = 0x20003c1f
```
How I analysis bug code?

thank you.

zherczeg, 2017-07-20:
The returned value is a jerry value, and cannot be printed by printf. To check whether an error is occured use the `jerry_value_has_error_flag` function. To check the type of the value use `jerry_value_is_...` functions. You can also use a `try-catch` block to capture the error, and print the error with print. However the best way to capture any error is using the debugger:

http://jerryscript.net/debugger/

It stops at the error position (unless it is a parse error), and you can inspect the source code, variables, etc.

LaszloLango, 2017-07-20:
My guess is that there is a typo in this line
`const jerry_char_t script[] = "var hl = \"hello\";var wo = \"world\";prin(hl + \" \" + wo);";`
I think there is no `prin` function registered maybe `print`. But the `print` is not a builtin function.

LaszloLango, 2017-07-20:
I just saw that you wrote `prin` is intentional.

LaszloLango, 2017-07-20:
The best way to debug is to use the debugger as @zherczeg mentioned it. If you cannot or don't want to use it then you can try to adapt the `print_unhandled_exception` function from [jerry-main/main-unix.c](https://github.com/jerryscript-project/jerryscript/blob/master/jerry-main/main-unix.c#L134)

akosthekiss, 2017-07-20:
If the code above is just an example and the issue actually appears in an undisclosed JS program, then you may use something like this in your C code to print `jerry_value_t`s:
```c
void
print_string_form (jerry_value_t val)
{
  jerry_value_t str_val = jerry_value_to_string (val);
  jerry_size_t str_size = jerry_get_string_size (str_val);

  jerry_char_t str_buf[str_size + 1];
  jerry_string_to_char_buffer (str_val, str_buf, str_size);
  err_str_buf[str_size] = 0;

  jerry_release_value (str_val);

  printf ("%s", str_buf);
}
```

But please note that your current `main` is not dealing with errors nicely. You should do something like:
```c
jerry_value_t parse_ret_val = jerry_parse (script, script_size, false);
if (!jerry_value_has_error_flag (parse_ret_val))
{
  jerry_value_t returned_value = jerry_run (parse_ret_val);
  if (jerry_value_has_error_flag (returned_value))
  {
    printf ("Run failed: ");
    jerry_value_clear_error_flag (&returned_value);
    print_string_form (returned_value);
    printf ("\n");
  }
  jerry_release_value (returned_value);
}
else
{
  printf ("Parse failed: ");
  jerry_value_clear_error_flag (&parse_ret_val);
  print_string_form (parse_ret_val);
  printf ("\n");
}
jerry_release_value (parse_ret_val);
```

You can get further inspiration from jerry-main.

To make sure that the printed errors are meaningful, enable error messages in the engine at build time (`./tools/build.py --error-messages=on`, or `cmake -DFEATURE_ERROR_MESSAGES=ON`, or `cc -DJERRY_ENABLE_ERROR_MESSAGES` -- depending on your build environment).

One more note: please don't expect helpful replies if you don't disclose detailed information about your issue. There are scenarios of course when the exact code that triggers problems cannot be shown but an as-complete-as-possible description of your environment and an as-minimal-as-possible reproduction case (not equivalent to the proprietary issue-triggering program!) is normally needed. I cannot see replies to the questions I've posted, which asked for such info.

k-harada413, 2017-07-24:
I get expected answer.
I try this way and debug my code.
Thank you  for being patient with my English and my technical skill.

closed 2017-07-24 by k-harada413