jerryscript-project/jerryscript #1934: build-in array shift/unshift error

topelinux, 2017-07-21:
run latest `jerry` with following test case, will encounter `Script Error: TypeError`, with `node` no error occur.

jerry is build by 

```bash
python tools/build.py
```
git header

```
d75710c `replace_str_curr_p` must be inspected before dereferencing (#1926)
ce1d555 Fix regression after #1927
```
test case

```js
 var devNum = 38;
 var devQueue = [];
 for (var i = 0; i < devNum; i++) {
     devQueue.push(i);
 }

 for (var j = 0; j < devNum; j++) {
     print(devQueue[0], devQueue.length);
     devQueue.shift();
     devQueue.unshift(1);
     devQueue.shift();
 }
```

LaszloLango, 2017-07-21:
Thanks for reporting. The bug is in the property hashmap implementation. Until we fix it you can disable the hashmap in the `jerry-core/config.h` by defining `CONFIG_ECMA_PROPERTY_HASHMAP_DISABLE` at line 71.

**Build command**

`tools/build.py --error-messages=on --debug`

**Output**

`ICE: Assertion 'ECMA_PROPERTY_IS_NAMED_PROPERTY (*property_p)' failed at /mnt/work/Samsung/jerryscript/jerry-core/ecma/base/ecma-property-hashmap.c(ecma_property_hashmap_find):510.`

**Backtrace**
```
#0  0x000000000046f33c in syscall_2 () at /mnt/work/Samsung/jerryscript/jerry-libc/target/posix/jerry-asm.S:59
#1  0x0000000000404b0e in raise (sig=6) at /mnt/work/Samsung/jerryscript/jerry-libc/target/posix/jerry-libc-target.c:95
#2  0x0000000000404ae0 in abort () at /mnt/work/Samsung/jerryscript/jerry-libc/target/posix/jerry-libc-target.c:81
#3  0x0000000000406f10 in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION)
    at /mnt/work/Samsung/jerryscript/jerry-port/default/default-fatal.c:71
#4  0x000000000042df5a in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION)
    at /mnt/work/Samsung/jerryscript/jerry-core/jrt/jrt-fatals.c:58
#5  0x000000000042dfaa in jerry_assert_fail (assertion=0x480a78 "ECMA_PROPERTY_IS_NAMED_PROPERTY (*property_p)", 
    file=0x4808b0 "/mnt/work/Samsung/jerryscript/jerry-core/ecma/base/ecma-property-hashmap.c", 
    function=0x483080 <__func__.3693.lto_priv.643> "ecma_property_hashmap_find", line=510)
    at /mnt/work/Samsung/jerryscript/jerry-core/jrt/jrt-fatals.c:82
#6  0x000000000046604b in ecma_property_hashmap_find (hashmap_p=0x694578 <jerry_global_heap+856>, 
    name_p=0x694650 <jerry_global_heap+1072>, property_real_name_cp=0x7fffffffd41e)
    at /mnt/work/Samsung/jerryscript/jerry-core/ecma/base/ecma-property-hashmap.c:510
#7  0x0000000000467036 in ecma_find_named_property (obj_p=0x694360 <jerry_global_heap+320>, 
    name_p=0x694650 <jerry_global_heap+1072>) at /mnt/work/Samsung/jerryscript/jerry-core/ecma/base/ecma-helpers.c:623
#8  0x000000000042b56e in ecma_op_object_put (object_p=0x694360 <jerry_global_heap+320>, 
    property_name_p=0x694650 <jerry_global_heap+1072>, value=592, is_throw=true)
    at /mnt/work/Samsung/jerryscript/jerry-core/ecma/operations/ecma-objects.c:782
#9  0x000000000044dbd5 in ecma_builtin_array_prototype_object_unshift.lto_priv.478 (this_arg=323, args=0x7fffffffd86c, 
    args_number=1) at /mnt/work/Samsung/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-array-prototype.c:1668
#10 0x000000000044fae9 in ecma_builtin_array_prototype_dispatch_routine (builtin_routine_id=45, this_arg_value=323, 
    arguments_list=0x7fffffffd86c, arguments_number=1)
    at /mnt/work/Samsung/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-array-prototype.inc.h:53
#11 0x0000000000456226 in ecma_builtin_dispatch_routine (builtin_object_id=ECMA_BUILTIN_ID_ARRAY_PROTOTYPE, builtin_routine_id=45, 
    this_arg_value=323, arguments_list=0x7fffffffd86c, arguments_number=1)
    at /mnt/work/Samsung/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:37
#12 0x00000000004566ad in ecma_builtin_dispatch_call (obj_p=0x694660 <jerry_global_heap+1088>, this_arg_value=323, 
    arguments_list_p=0x7fffffffd86c, arguments_list_len=1)
    at /mnt/work/Samsung/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.c:844
#13 0x0000000000432277 in ecma_op_function_call (func_obj_p=0x694660 <jerry_global_heap+1088>, this_arg_value=323, 
    arguments_list_p=0x7fffffffd86c, arguments_list_len=1)
    at /mnt/work/Samsung/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:458
#14 0x000000000041ffa4 in opfunc_call.lto_priv.129 (frame_ctx_p=0x7fffffffd8b0)
    at /mnt/work/Samsung/jerryscript/jerry-core/vm/vm.c:411
#15 0x0000000000415501 in vm_execute (frame_ctx_p=0x7fffffffd8b0, arg_p=0x0, arg_list_len=0)
    at /mnt/work/Samsung/jerryscript/jerry-core/vm/vm.c:2746
#16 0x0000000000415782 in vm_run (bytecode_header_p=0x694508 <jerry_global_heap+744>, this_binding_value=27, 
    lex_env_p=0x694248 <jerry_global_heap+40>, is_eval_code=false, arg_list_p=0x0, arg_list_len=0)
    at /mnt/work/Samsung/jerryscript/jerry-core/vm/vm.c:2826
#17 0x000000000041fbf8 in vm_run_global (bytecode_p=0x694508 <jerry_global_heap+744>)
    at /mnt/work/Samsung/jerryscript/jerry-core/vm/vm.c:231
#18 0x000000000046bf1a in jerry_run (func_val=251) at /mnt/work/Samsung/jerryscript/jerry-core/api/jerry.c:409
#19 0x000000000046ea70 in main (argc=3, argv=0x7fffffffdb98) at /mnt/work/Samsung/jerryscript/jerry-main/main-unix.c:695
```

rerobika, 2017-07-25:
I'll check it!

PR #1938 (2017-08-02): ecma_delete_property must recreate the hashmap even if one of the properties is still valid
Fixes #1934.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
commit a3885be6 (2017-08-02): ecma_delete_property must recreate the hashmap even if one of the properties is still valid. (#1938)

Fixes #1934.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
closed 2017-08-02 by zherczeg