jerryscript-project/jerryscript #1947: in jmem_heap_finalize

rerobika, 2017-08-01:
###### Jerry version:

```
Checked revision: 6fe4d06
Build command: ./tools/build.py --clean --debug --profile=es2015-subset --compile-flag=-m32 --system-allocator=on --jerry-libc=off --error-messages=on
```
###### OS:

```
Ubuntu 16.04.2 LTS
```
###### Test case:

```javascript
function $ ( ) { function $ ( ) { switch ( $ ? $ : \u018E ) {} } }
```

###### Backtrace:

```
ICE: Assertion 'JERRY_CONTEXT (jmem_heap_allocated_size) == 0' failed at jerryscript//jerry-core/jmem/jmem-heap.c(jmem_heap_finalize):184.
Error: ERR_FAILED_INTERNAL_ASSERTION

```

<sup>Found by  [Fuzzinator](http://fuzzinator.readthedocs.io/) with [Grammarinator](https://github.com/renatahodovan/grammarinator)</sup>

akosthekiss, 2017-08-07:
The issue is confirmed to be valid on latest master c242248b (the hash 6fe4d06 does not seem to be in the master branch) on macOS 10.12.5, too. Moreover, the issue is not ES2015-specific, it can be reproduced by building jerry simply with `tools/build.py --clean --debug --error-messages=on`. The error message is as in the original report, the backtrace is as follows (as reported by `lldb -- ./build/bin/jerry --abort-on-fail issue-1947.js`):
```
    frame #5: 0x0000000100006f9e jerry`jerry_assert_fail(assertion="JERRY_CONTEXT (jmem_heap_allocated_size) == 0", file="jerry-core/jmem/jmem-heap.c", function="jmem_heap_finalize", line=184) at jrt-fatals.c:82
    frame #6: 0x0000000100097eb6 jerry`jmem_heap_finalize at jmem-heap.c:184
    frame #7: 0x00000001000039fe jerry`jmem_finalize at jmem-allocator.c:51
    frame #8: 0x0000000100003531 jerry`jerry_cleanup at jerry.c:180
    frame #9: 0x0000000100001dd6 jerry`main(argc=3, argv=0x00007fff5fbffa70) at main-unix.c:823
```

DanielBallaSZTE, 2017-08-24:
I've started working on it.

commit a51def40 (2017-09-07): Fix #1947

If a literal was assigned the unused flag it wasn't freed, however it could have been not empty, therefore should've been freed.

JerryScript-DCO-1.0-Signed-off-by: Daniel Balla dballa@inf.u-szeged.hu
closed 2017-09-07 by yichoi