jerryscript-project/jerryscript #1949: JSON.parse/stringify C APIs?

martijnthe, 2017-08-01:
I think it would be nice if the `jerry-core.h` API would have C APIs that match JSON.parse/stringify.

Proposal API:

```
typedef jerry_value_t (*jerry_json_reviver_t) (jerry_value_t key, jerry_value_t value);

jerry_value_t
jerry_json_parse_utf8 (const jerry_char_t *utf8_str,
                       size_t str_sz,
                       jerry_json_reviver_t reviver);

typedef jerry_value_t (*jerry_json_replacer_t) (jerry_value_t key, jerry_value_t value);

jerry_value_t
jerry_json_stringify(jerry_value_t value,
                     jerry_json_replacer_t replacer,
                     jerry_value_t space);
```

Note the output of `jerry_json_stringify` is not a UTF-8 C string (`jerry_char_t[]`). Although in most cases you probably want a C string...:

1. how to allocate the buffer for the C string?
2. what's an elegant way to deal with any errors – perhaps a `jerry_value_t *error_out`?

Thoughts?

martijnthe, 2017-08-01:
cc @jiangzidong

jiangzidong, 2017-08-02:
> 1. how to allocate the buffer for the C string?
We can pass the allocated buf into the jerry_json_stringify, if the length is not big enough, return error.
or we pass the "malloc" function into the `jerry_json_stringify`, but this will make the API a little harder to use.

> 2. what's an elegant way to deal with any errors
Why not just return the error, like `jerry_call_function` did. if the ret has error flag, then it is the error.
It is the common way used in jerry-api.

zherczeg, 2017-08-02:
1) I think the system constructs a string value, so I am not sure passing a buffer would work.
2) Yes, the returned value can have an error flag set.