jerryscript-project/jerryscript #1952: Remove stringification of JERRYX_NATIVE_MODULE module_name argument?

martijnthe, 2017-08-03:
See https://github.com/jerryscript-project/jerryscript/blob/master/jerry-ext/include/jerryscript-ext/module.h#L51

`module_name` is stringified, which will cause the argument to not be evaluated by the preprocessor but used verbatim.

After having used the macro in practice, I found that it's not desirable to stringify it, because:

- it prevents one from creating macros that call the `JERRYX_NATIVE_MODULE` macro and pass in a variable module name

- it makes it impossible to single-source the module name, i.e. I cannot write something like:
```
#define MY_MODULE_NAME my_module
#define MY_MODULE_NAME_STR STRINGIFY(MY_MODULE_NAME)

void
foo (void)
{
  const jerry_value_t m = call_require(MY_MODULE_NAME_STR);
  // ...
}

// module will actually be named "MY_MODULE_NAME"...
JERRYX_NATIVE_MODULE(MY_MODULE_NAME, ...)   
```

Therefore I propose to remove the stringification (`#`) from the macro.
I realize it's a breaking change, but the API is pretty new and the migration path is super simple.

Thoughts @gabrielschulhof ?

cc @jiangzidong

jiangzidong, 2017-08-11:
@martijnthe I agree with you. And I think we can "break" the API.

gabrielschulhof, 2017-09-17:
@martijnthe @jiangzidong I'm working on this to help out Tim Harvey. I'll have something in a bit.

zherczeg, 2017-09-18:
Ok for me.

gabrielschulhof, 2017-09-20:
@martijnthe I found a [solution](https://gcc.gnu.org/onlinedocs/cpp/Stringizing.html#Stringizing) which will allow both the use of `#` in the macro implementation and passing in a macro that evaluates to a non-quoted string.

Basically, if we do
```c
#define JERRYX_NATIVE_MODULE(module_name, on_resolve_cb) \
  JERRYX_NATIVE_MODULE_IMPLEM(module_name, on_resolve_cb)

#define JERRYX_NATIVE_MODULE_IMPLEM(module_name, on_resolve_cb) \
/* actual implementation of macro */
```
Then we can use it like so:
```c
#define MY_MODULE_NAME my_module_name

static jerry_value_t Init (void) {return jerry_create_number (42);}

JERRYX_NATIVE_MODULE(MY_MODULE_NAME, Init)
```

The key is to have a macro call a macro.

gabrielschulhof, 2017-09-20:
This is what Node.js does too.

martijnthe, 2017-09-20:
@gabrielschulhof yep, thanks.

closed 2017-09-22 by zherczeg