jerryscript-project/jerryscript #1961: Request: API to get current ref count

grgustaf, 2017-08-11:
Apologies if there is some way to do this that I have missed. But it seems like it would be quite useful for debugging to have a method like:

uint16_t jerry_get_ref_count(jerry_value_t obj);

I've managed to create a macro that hacks my way to this information but it will likely break with future changes to JrS and probably only works for my specific build conditions.

martijnthe, 2017-08-11:
+1 I've also wished for this a couple times.

I think it would also be useful to have a GDB (or LLDB?) command library (for example using the Python GDB API) for common things like eval'ing a string from GDB, printing the string value of a `jerry_value_t`, dumping the JrS heap, etc.

jiangzidong, 2017-08-12:
+1, and

wille-io, 2018-06-06:
+1, a necessary helper for finding non-released values faster
@grgustaf would you please share your macro?

grgustaf, 2018-06-06:
@wille-io Don't say I didn't warn you: [nasty hack here](https://github.com/intel/zephyr.js/commit/7762a50300c1b26b2a4de11daf9470f7da814a60) :)

This was just making use of where the reference count was stored relative to the JerryScript object pointer, in my build conditions. Haven't used it for quite a while.