jerryscript-project/jerryscript #1962: ES2015 or ES2017 for new ArrayBuffer(length)

glistening, 2017-08-11:
The definition of `new ArrayBuffer(length)` is changed in ES2017.

For example,

ES2015
new ArrayBuffer(1.1) should throw RangeError.

ES2017
new ArrayBuffer(1.1) creates ArrayBuffer with length 1.

I would like to follow ES2017's definition.

Major engines already work like ES2017 spec. ( Or ES specification may be updated to follow major engines.  [ES2016](https://www.ecma-international.org/ecma-262/7.0/index.html#sec-arraybuffer-length) specification was same to ES2015. ).

jiangzidong, 2017-08-12:
I prefer following ES2017 if there are some conflict with ES2015, since that is major engines' choice.

Maybe we should change the profile name `es2015-subset` to `es2017-subset` ?

glistening, 2017-08-16:
@jiangzidong Thank you for opinion. I also thought that `es2015-subset` is correct name if we follow latest specification like new ArrayBuffer(length). However, ES specification will be changed for several forthcoming years ( es2018, es2019, ... ), and there may be feature X, Y,  that major specification follows es2015 yet. I cannot find good name that covers this situation. So I would like to keep the current name - `es2015-subset`.

jiangzidong, 2017-08-16:
Yes, make sense.

closed 2017-08-17 by glistening