jerryscript-project/jerryscript #1965: Under gcc v7.1.0, the FALLTHRU comments to be placed differently

cnnblike, 2017-08-13:
Hi, when I was trying to build jerryscript under arm-none-eabi-gcc, I found that the FALLTHRU comments need to placed after right brackets.
So here is the interesting part: I found a minor fix on this, but found it can't pass the vera++ check (which forced us to place FALLTHRU before right bracket).

LaszloLango, 2017-08-21:
What version of JerryScript are you using? #1878 should solve this issue. Please check the latest master if you are working with an older version.

galpeter, 2017-08-23:
I've tried out the current master with the arm-non-eabi-gcc v7.1 (toolchain file: cmake/toolchain_mcu_stm32f4.cmake) and there was no error report.
The PR mentioned above should disable the warning report till we improve the vera++ checker.

closed 2017-08-24 by cnnblike