jerryscript-project/jerryscript #1973: Hang while parsing string exponent to number (004)

corporateshill, 2017-08-18:
Seemingly infinite loop while trying to convert a string representing an exponential number to a number.

On Ubuntu 16.04.2,

  $ python tools/build.py --compile-flag=-m32 --clean --jerry-libc=OFF --system-allocator ON --compile-flag=-ggdb3 --debug --link-lib mcheck

  $ cat hang.js
  777E7777777777-----00,');

  $ MALLOC_CHECK_=1 gdb -q -x mcheck.gdb --args ./build/bin/jerry hang.js
  Reading symbols from ./build/bin/jerry...done.
  Breakpoint 1 at 0x80ac9a8: file /afl/jerryscript/jerry-main/main-unix.c, line 410.

  Breakpoint 1, main (argc=0x2, argv=0xffffd644) at /afl/jerryscript/jerry-main/main-unix.c:410
  410 {
  $1 = 0x0
  ^C
  Program received signal SIGINT, Interrupt.
  0x080a049f in ecma_utf8_string_to_number (str_p=0x8178240 <buffer.lto_priv> "777E7777777777-----00,');\n", str_size=0xe) at /afl/jerryscript/jerry-core/ecma/base/ecma-helpers-conversion.c:694
  694       ECMA_NUMBER_CONVERSION_128BIT_INTEGER_DIV_10 (fraction_uint128);
  (gdb) shell date
  Fri Aug 18 13:57:58 PDT 2017
  (gdb) c
  Continuing.
  ^C
  Program received signal SIGINT, Interrupt.
  0x080a1097 in ecma_utf8_string_to_number (str_p=0x8178240 <buffer.lto_priv> "777E7777777777-----00,');\n", str_size=0xe) at /afl/jerryscript/jerry-core/ecma/base/ecma-helpers-conversion.c:694
  694       ECMA_NUMBER_CONVERSION_128BIT_INTEGER_DIV_10 (fraction_uint128);
  (gdb) shell date
  Fri Aug 18 13:58:31 PDT 2017
  (gdb)

LaszloLango, 2017-08-30:
This is only a performance issue. It does not stuck in the loop.

```bash
$> time build/bin/jerry hang.js 

real	1m6.096s
user	1m6.084s
sys	0m0.016s
```

rerobika, 2017-09-06:
I'll check it

closed 2017-10-27 by LaszloLango