jerryscript-project/jerryscript #1977: Add jerryscript-snapshot-compiler.js ?

martijnthe, 2017-08-23:
I'm interested in having a JavaScript library / module that can take a JS source string and produce a JerryScript snapshot. At a previous company, we had built something like this, simply by using Emscripten to transpile the C code that produces snapshots into JS. The advantage is that such a library is very easily "portable" to other platforms (i.e. cloud, mobile, browser, etc.)

Any objections if I add an option to JerryScript's build system to produce such a library?

martijnthe, 2017-08-23:
... also raises the question: how to test that the output is correct? I looked for tests in the codebase that check that the "binary" that `jerry_parse_and_save_snapshot` produces, matches an expected blob, but to my surprise I did not find such tests.

zherczeg, 2017-08-23:
Testing: we generate the snapshot version of each js test and run them. Their behavior must match the non-snapshot version.

martijnthe, 2017-08-23:
Gotcha. I was thinking the "snapshot format" is an ABI/binary contract, but I guess it's not really because it depends on how the various (non-packed) structs are laid out in memory, which is in theory a compiler specific implementation detail... correct?

I'm mostly interested in knowing whether snapshots produced by a Emscripten-transpiled compiler.js can be correctly executed by a JerryScript that is compiled by GCC. Hence my idea of comparing the output blob with an expected blob (produced by GCC-built JerryScript). Any thoughts about this?

zherczeg, 2017-08-24:
The C struct alignment is defined by the standard, and we use uint... types, So no, the snapshot format should not affected by the compiler except data alignment which is CPU specific. The 8 byte double format is also the same on all major CPU families.

A compiler.js should be able to produce the correct snapshot. Creating a correct double value could be a challenge though.

martijnthe, 2017-08-24:
> So no, the snapshot format should not affected by the compiler except data alignment which is CPU specific.

Emscripten emulates a 32-bit machine, so I guess as long as the target for which you are creating the snapshot is also 32-bit, it will work? Or are there more parameters to how data alignment is decided other than the machine's word size?

I wonder if it would make sense to add some info in the `jerry_snapshot_header_t` snapshot header about this and check that it matches before trying to run it?

> Creating a correct double value could be a challenge though.

Why is this do you think?

zherczeg, 2017-08-25:
Yes, a 32 bit machine should work without issues. Generating the byte code is not that big deal. Typed arrays could also be a great help (and you don't need emscripten).

> Why is this do you think?

IEEE compatible floating point numbers. You need a conversion library.

rerobika, 2019-10-25:
IMO this issue corresponds to #1901 which has already been closed. Closing the issue due to inactivity please reopen it if needed.

closed 2019-10-25 by rerobika