jerryscript-project/jerryscript #1986: Nominating SangGyu Lee (glistening) to be a JerryScript committer

zherczeg, 2017-08-29:
Hi,

I would like to nominate SangGyu Lee (glistening) to be a JerryScript committer. I trust him to follow our process. His patch list follows:

- Fix new ArrayBuffer(length) for variaous input (#1959)
- Implement typedarary.set(array, offset) (#1958)
- Change tizenrt-artik05x port reference board to artik053 (#1904)
- Fix build warning in jerry-ext/arg on tizenrt-artik05x (#1906)
- Don't allow supplementary character as identifier start or part (#1899)
- Handle negative zero in number multiplication (#1856)
- Add `setjmp` and `longjmp` functions for tizenrt-artik05x (#1825)
- Use external handlers in tizenrt-artik05x target (#1822)
- Include jerry-ext/handler only in nuttx-stm32f4 target (#1816)
- target: tizenrt-artik05x: Add missing jerry_port.c (#1786)
- target: tizenrt-artik05x: Initial porting (#1775)
- Fix maybe-uninitialized warning in ecma-objects.c (#1773)

yichoi, 2017-09-05:
Closing. No significant objection.

closed 2017-09-05 by yichoi

glistening, 2017-09-05:
Thank you! See you at workshop!