jerryscript-project/jerryscript #1993: Segmentation fault in ecma_builtin_number_prototype_object_to_fixed

renatahodovan, 2017-09-04:
###### Jerry version:

```
Checked revision: ce18704
Build command: ./tools/build.py --clean --debug
```
###### OS:

```
Ubuntu 16.04.3 LTS
```
###### Test case:

```javascript
parseFloat(Number.MAX_VALUE).toFixed(5);
```

###### Backtrace:

```
Program received signal SIGSEGV, Segmentation fault.
0x0000000000447bc3 in ecma_builtin_number_prototype_object_to_fixed (this_arg=393, arg=80)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-number-prototype.c:648
648 } /* ecma_builtin_number_prototype_object_to_fixed */
bt
#0  0x0000000000447bc3 in ecma_builtin_number_prototype_object_to_fixed (this_arg=393, arg=80) at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-number-prototype.c:648
#1  0x3834323134383136 in ?? ()
#2  0x0000003836333835 in ?? ()
#3  0x000000010096c0f0 in ?? ()
#4  0x00007fffffffcd2c in ?? ()
#5  0x0000003d00000189 in ?? ()
#6  0x00007fffffffcb20 in ?? ()
#7  0x000000000045e84e in ecma_builtin_dispatch_routine (builtin_object_id=<error reading variable: Cannot access memory at address 0x323034303430352e>, builtin_routine_id=<error reading variable: Cannot access memory at address 0x323034303430352a>, this_arg_value=<error reading variable: Cannot access memory at address 0x3230343034303526>, arguments_list=<error reading variable: Cannot access memory at address 0x323034303430351a>, arguments_number=<error reading variable: Cannot access memory at address 0x3230343034303522>) at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:91
Backtrace stopped: Cannot access memory at address 0x323034303430353a
```

<sup>Found by [Fuzzinator](http://fuzzinator.readthedocs.io/)</sup>

rerobika, 2017-09-04:
I'll check it.

akosthekiss, 2017-11-10:
Fixed by #1994

closed 2017-11-10 by akosthekiss