jerryscript-project/jerryscript #2023: what happen when jerry_call_function called in other thread

xfeli, 2017-09-25:
I see IoT.js use jerry_call_function in other thread not jerryscript thread, it is a right implementation? 
I test that code and no error happen, but it lead to javascript code run in multiple thread, it is dangerous?

https://github.com/Samsung/iotjs/blob/master/src/platform/linux/iotjs_module_gpio-linux.c
line: 86 (iotjs_jhelper_call_ok is wrapper for jerry_call_function)

zherczeg, 2017-09-26:
Could be dangerous. Are you sure it is in the other thread?

xfeli, 2017-09-27:
@zherczeg Yes, I call gettid() in `change event` in javascript code, it's really other thread.

zherczeg, 2017-09-27:
I think the concept of iot.js (and node.js) is that only the main thread execute js, while worker threads do the io. This should also be moved back to the main thread.

akosthekiss, 2017-10-11:
That part of the original question which is related to the engine (whether multiple threads could/should call JerryScript API functions on the same engine context) has been answered (should not). However, the IoT.js-related observations should be discussed in the issue tracker of that project. Please, open a new issue at https://github.com/Samsung/iotjs/issues (preferably cross-referencing to this one) and close this thread.

akosthekiss, 2017-10-30:
Has this issue been reported against (or already fixed in) IoT.js? Can we close this here?

closed 2018-04-01 by xfeli