jerryscript-project/jerryscript #2029: Does jerryscipt support Websockets JS library?

Ash200012, 2017-09-28:
I am trying to take my javascript code (previously run in a browser) and run it from jerryscript. My JS code uses websockets libraries. Does jerryscript support this or any other network protocol library?

zherczeg, 2017-09-29:
JerryScript is a JS engine, not an environment like node.js. There are several other projects, such as https://github.com/Samsung/iotjs which provides network communication on the top of jerryscript.

Ash200012, 2017-09-29:
Thank you for your reply. IoTjs seems to have limited platform support..are there any other projects on top of jerryscript or not that have network communication support?

zherczeg, 2017-09-29:
Maybe MBED or Zephyr has them.

zherczeg, 2017-09-29:
Btw what is your platform?

Ash200012, 2017-09-29:
ARM boxes and potentially MIPS. I am looking for a very low footprint JS engine where I can write a JS websocket or http server that can do things like discovery, send/receive messages in JSON format, and even do downloading of files.

zherczeg, 2017-09-29:
I mean your OS

Ash200012, 2017-09-29:
Many flavors of custom linux..busybox linux, etc. python may not be available on them

zherczeg, 2017-09-29:
IoT.js runs on any Linux (only a basic libc required), no need any python (except for compiling, but you can cross compile on desktop).

Ash200012, 2017-09-29:
OK thanks I will look into it if it is my only option.

akosthekiss, 2017-10-11:
Closing as WebSockets is out of scope for the engine and discussion is continued at the issue tracker of IoT.js (https://github.com/Samsung/iotjs/issues/1241).

closed 2017-10-11 by akosthekiss