jerryscript-project/jerryscript #2030: mbed debugger error

RozenAstrayChen, 2017-09-29:
I want to use mbed-cli debugger,so I edit `/mbed-js-example/build/jerryscript/targets/mbedos5` ->`DEBUG?=1` 
but I got this error
`Compile [100.0%]: test_env.cpp
Link: mbedos5
./BUILD/NUCLEO_F401RE/GCC_ARM/mbed-os/rtos/mbed_boot.o: In function `__wrap_main':
/Users/Rozen_mac/code/node/mbed-js-example/build/jerryscript/targets/mbedos5/./mbed-os/rtos/mbed_boot.c:480: undefined reference to `main'
collect2: error: ld returned 1 exit status
[ERROR] ./BUILD/NUCLEO_F401RE/GCC_ARM/mbed-os/rtos/mbed_boot.o: In function `__wrap_main':
/Users/Rozen_mac/code/node/mbed-js-example/build/jerryscript/targets/mbedos5/./mbed-os/rtos/mbed_boot.c:480: undefined reference to `main'
collect2: error: ld returned 1 exit status

[mbed] ERROR: "/Users/Rozen_mac/anaconda/bin/python" returned error code 1.
[mbed] ERROR: Command "/Users/Rozen_mac/anaconda/bin/python -u /Users/Rozen_mac/code/node/mbed-js-example/build/jerryscript/targets/mbedos5/mbed-os/tools/make.py -t GCC_ARM -m NUCLEO_F401RE --profile ./mbed-os/tools/profiles/debug.json --source . --source ../../ --build ./BUILD/NUCLEO_F401RE/GCC_ARM -j0 -D CONFIG_MEM_HEAP_AREA_SIZE=(1024*16)" in "/Users/Rozen_mac/code/node/mbed-js-example/build/jerryscript/targets/mbedos5"
---
make: *** [all] Error 1
`

akosthekiss, 2017-10-11:
I cannot comment on the `throw er;` part but the start of the error log seems to be clear about the linked binary not fitting into the flash of your target device ("section .text will not fit in region FLASH" and "region FLASH overflowed by 81380 bytes"). A note though: 81K overflow is a lot. What was the size of the text section with DEBUG=0, and how large is the flash on your device?

RozenAstrayChen, 2017-10-15:
My device is  STM32 NUCLEO F401RE,flash size is ==512KB==

akosthekiss, 2017-10-15:
And what was the size of the program before debug was enabled?

RozenAstrayChen, 2017-10-16:
How I can find the size of program before debug was enabled? I'm not setting this

akosthekiss, 2017-10-16:
I don't see a problem with the linker script (at first glance). As to your question: I use mbed cli for building my project (e.g., `mbed compile --clean`), there I get something like this as the last lines of console output:
```
Total Static RAM memory (data + bss): 16468 bytes
Total Flash memory (text + data): 256063 bytes
```
Do you have anything similar?

RozenAstrayChen, 2017-10-16:
Ya I find this message when i build project 
```
Total Static RAM memory (data + bss): 31152 bytes
Total RAM memory (data + bss + heap + stack): 31152 bytes
Total Flash memory (text + data + misc): 375513 bytes
```

akosthekiss, 2017-10-16:
Okay, I think that you'll have to do without DEBUG=1 for now. I tried a debug build with `mbed compile --clean --profile debug` on my own project and I also got a `region 'FLASH' overflowed by 9248 bytes` error although my starting release size was smaller than yours (and I have the same flash size). I have no idea yet what exactly blows the code to twice the size (but `-O0` for debug vs `-Os` for release  certainly doesn't help).

RozenAstrayChen, 2017-10-16:
Ok Thanks you for your reply. :)

akosthekiss, 2017-10-16:
Closing issue as it's not a bug (neither in mbed-os nor in jerryscript) but simply a debug build not fitting into the flash of a specific device. To be solved by specifying a custom set of compiler flags (e.g., `-Os -g3`) or by defining a custom build profile (https://docs.mbed.com/docs/mbed-os-handbook/en/latest/dev_tools/build_profiles/).

closed 2017-10-16 by akosthekiss