jerryscript-project/jerryscript #2031: New error management system for JerryScript

zherczeg, 2017-09-29:
During the meeting it was discussed that other engines use an "error mode" rather than signal the error as a flag in the return value. The advantage of that approach is we can free up the flag bit in the jerry value, so new JS types such as symbols could be implemented. The drawback is a breaking API change. What shall we do?

LaszloLango, 2017-09-29:
I'd leave the current approach until we don't need that bit. But if decide to implement the symbols now, then we can change to this "error mode". Maybe we should deploy a new release before a significant API change like this.

akosthekiss, 2017-10-11:
The bare fact alone that other engines use a global error flag/mode is not convincing enough to me to do such a heavy API change. There is quite some code around JerryScript now that will break and will need **heavy** manual follow-up update work to get things building and running again.

So, I think that first we need to evaluate whether global error mode is the only way to support symbols or there are other approaches that are not so API breaking.

zherczeg, 2017-10-12:
No problem. #2042 already start doing this without API change. So we can postpone the api change to 2.0. But is worth discussing it.

martijnthe, 2017-11-10:
What's the time frame for 2.0?

akosthekiss, 2018-05-10:
@zherczeg Is this still an ongoing topic or is it closed with the merging of #2042?

akosthekiss, 2018-05-10:
For the records: the follow-up issue is #2318