jerryscript-project/jerryscript #2038: Problem with generate_pins.py script in mbedos5

syed-zeeshan, 2017-10-04:
I am trying to build my project for STM32F429ZI and I have noticed an issue with generate_pins.py script. It fails while trying to evaluate expressions. For example, I am getting this error while building.

**zMacBookPro:mbed-js-example zeeshan$ gulp --target=NUCLEO_F429ZI
[15:30:03] Using gulpfile ~/Programming/mbed/mbed-js-example/gulpfile.js
[15:30:03] Starting 'make-build-dir'...
[15:30:03] Finished 'make-build-dir' after 149 μs
[15:30:03] Starting 'makefile'...
[15:30:03] Starting 'bundle'...
[15:30:03] Starting 'ignorefile'...
[15:30:03] Finished 'makefile' after 50 ms
[15:30:03] Starting 'get-jerryscript'...
[15:30:03] Finished 'get-jerryscript' after 198 μs
[15:30:03] Starting 'getlibs'...
[15:30:03] Finished 'ignorefile' after 42 ms
[15:30:03] Finished 'getlibs' after 71 ms
$ make getlibs
make: Nothing to be done for `getlibs'.
[15:30:03] Finished 'bundle' after 138 ms
[15:30:03] Starting 'cppify'...
[15:30:03] Finished 'cppify' after 99 ms
[15:30:03] Starting 'build'...
[15:30:04] Found no native packages.
$ make BOARD=NUCLEO_F429ZI EXTRAS=
[15:30:04] build/make
[15:30:04] Finished 'build' after 1.44 s
[15:30:04] Starting 'default'...
[15:30:04] Finished 'default' after 42 μs
cd jerryscript/targets/mbedos5 && pip install -r tools/requirements.txt && rm -f source/js_encoded.cpp && rm -f source/pins.cpp && rm -f source/main.cpp && make BOARD=NUCLEO_F429ZI EXTRA_SRC="../../../source" EXTERN_BUILD_DIR=../../../out/NUCLEO_F429ZI NO_JS=1
Requirement already satisfied: pycparser<=2.17 in /Library/Python/2.7/site-packages (from -r tools/requirements.txt (line 1))
Requirement already satisfied: simpleeval in /Library/Python/2.7/site-packages (from -r tools/requirements.txt (line 2))
Requirement already satisfied: pycparserext in /Library/Python/2.7/site-packages (from -r tools/requirements.txt (line 3))
Requirement already satisfied: ply>=3.4 in /Library/Python/2.7/site-packages (from pycparserext->-r tools/requirements.txt (line 3))
python tools/generate_pins.py NUCLEO_F429ZI
Traceback (most recent call last):
  File "tools/generate_pins.py", line 249, in <module>
    main()
  File "tools/generate_pins.py", line 239, in main
    pins = enumerate_pins(pins_file, ['./tools'] + list(includes), defines)
  File "tools/generate_pins.py", line 160, in enumerate_pins
    return visitor.visit(parsed_ast)
  File "tools/generate_pins.py", line 107, in visit
    value = value or self.visit(child_node)
  File "tools/generate_pins.py", line 107, in visit
    value = value or self.visit(child_node)
  File "tools/generate_pins.py", line 103, in visit
    value = self.visit_typedecl(node)
  File "tools/generate_pins.py", line 135, in visit_typedecl
    pins[pin.name] = evaluator.eval(expr.strip())
  File "/Library/Python/2.7/site-packages/simpleeval.py", line 274, in eval
    return self._eval(ast.parse(expr.strip()).body[0].value)
  File "/Library/Python/2.7/site-packages/simpleeval.py", line 285, in _eval
    return handler(node)
  File "/Library/Python/2.7/site-packages/simpleeval.py", line 307, in _eval_binop
    return self.operators[type(node.op)](self._eval(node.left),
  File "/Library/Python/2.7/site-packages/simpleeval.py", line 285, in _eval
    return handler(node)
  File "/Library/Python/2.7/site-packages/simpleeval.py", line 378, in _eval_name
    raise NameNotDefined(node.id, self.expr)
simpleeval.NameNotDefined: 'PA_0' is not defined for expression 'PA_0 | ALT0'
make[1]: *** [source/pins.cpp] Error 1
make: *** [all] Error 2
events.js:182
      throw er; // Unhandled 'error' event
      ^

Error: Command `make BOARD=NUCLEO_F429ZI EXTRAS=` exited with code 2
    at ChildProcess.handleSubShellExit (/Users/zeeshan/Programming/mbed/mbed-js-example/node_modules/gulp-run/command.js:166:13)
    at Object.onceWrapper (events.js:318:30)
    at emitTwo (events.js:125:13)
    at ChildProcess.emit (events.js:213:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)**

I have also tried building for REALTEK_RTL8195AM and I am getting similar errors:

**zMacBookPro:mbed-js-example zeeshan$ gulp --target=REALTEK_RTL8195AM
[15:38:37] Using gulpfile ~/Programming/mbed/mbed-js-example/gulpfile.js
[15:38:37] Starting 'make-build-dir'...
[15:38:38] Finished 'make-build-dir' after 164 μs
[15:38:38] Starting 'makefile'...
[15:38:38] Starting 'bundle'...
[15:38:38] Starting 'ignorefile'...
[15:38:38] Finished 'makefile' after 40 ms
[15:38:38] Starting 'get-jerryscript'...
[15:38:38] Finished 'get-jerryscript' after 115 μs
[15:38:38] Starting 'getlibs'...
[15:38:38] Finished 'ignorefile' after 42 ms
[15:38:38] Finished 'getlibs' after 75 ms
$ make getlibs
make: Nothing to be done for `getlibs'.
[15:38:38] Finished 'bundle' after 130 ms
[15:38:38] Starting 'cppify'...
[15:38:38] Finished 'cppify' after 92 ms
[15:38:38] Starting 'build'...
[15:38:39] Found no native packages.
$ make BOARD=REALTEK_RTL8195AM EXTRAS=
[15:38:39] build/make
[15:38:39] Finished 'build' after 1.52 s
[15:38:39] Starting 'default'...
[15:38:39] Finished 'default' after 43 μs
cd jerryscript/targets/mbedos5 && pip install -r tools/requirements.txt && rm -f source/js_encoded.cpp && rm -f source/pins.cpp && rm -f source/main.cpp && make BOARD=REALTEK_RTL8195AM EXTRA_SRC="../../../source" EXTERN_BUILD_DIR=../../../out/REALTEK_RTL8195AM NO_JS=1
Requirement already satisfied: pycparser<=2.17 in /Library/Python/2.7/site-packages (from -r tools/requirements.txt (line 1))
Requirement already satisfied: simpleeval in /Library/Python/2.7/site-packages (from -r tools/requirements.txt (line 2))
Requirement already satisfied: pycparserext in /Library/Python/2.7/site-packages (from -r tools/requirements.txt (line 3))
Requirement already satisfied: ply>=3.4 in /Library/Python/2.7/site-packages (from pycparserext->-r tools/requirements.txt (line 3))
python tools/generate_pins.py REALTEK_RTL8195AM
Traceback (most recent call last):
  File "tools/generate_pins.py", line 249, in <module>
    main()
  File "tools/generate_pins.py", line 239, in main
    pins = enumerate_pins(pins_file, ['./tools'] + list(includes), defines)
  File "tools/generate_pins.py", line 160, in enumerate_pins
    return visitor.visit(parsed_ast)
  File "tools/generate_pins.py", line 107, in visit
    value = value or self.visit(child_node)
  File "tools/generate_pins.py", line 107, in visit
    value = value or self.visit(child_node)
  File "tools/generate_pins.py", line 103, in visit
    value = self.visit_typedecl(node)
  File "tools/generate_pins.py", line 135, in visit_typedecl
    pins[pin.name] = evaluator.eval(expr.strip())
  File "/Library/Python/2.7/site-packages/simpleeval.py", line 274, in eval
    return self._eval(ast.parse(expr.strip()).body[0].value)
  File "/Library/Python/2.7/site-packages/simpleeval.py", line 285, in _eval
    return handler(node)
  File "/Library/Python/2.7/site-packages/simpleeval.py", line 307, in _eval_binop
    return self.operators[type(node.op)](self._eval(node.left),
  File "/Library/Python/2.7/site-packages/simpleeval.py", line 285, in _eval
    return handler(node)
  File "/Library/Python/2.7/site-packages/simpleeval.py", line 307, in _eval_binop
    return self.operators[type(node.op)](self._eval(node.left),
  File "/Library/Python/2.7/site-packages/simpleeval.py", line 285, in _eval
    return handler(node)
  File "/Library/Python/2.7/site-packages/simpleeval.py", line 378, in _eval_name
    raise NameNotDefined(node.id, self.expr)
simpleeval.NameNotDefined: 'PORT_A' is not defined for expression '(PORT_A << 4) | 0'
make[1]: *** [source/pins.cpp] Error 1
make: *** [all] Error 2
events.js:182
      throw er; // Unhandled 'error' event
      ^

Error: Command `make BOARD=REALTEK_RTL8195AM EXTRAS=` exited with code 2
    at ChildProcess.handleSubShellExit (/Users/zeeshan/Programming/mbed/mbed-js-example/node_modules/gulp-run/command.js:166:13)
    at Object.onceWrapper (events.js:318:30)
    at emitTwo (events.js:125:13)
    at ChildProcess.emit (events.js:213:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)**

For the time being, I am fixing my issue with STM32F429ZI by writing the following line in try statement:

                    try:
                        **pins[pin.name] = evaluator.eval(expr.strip())**
                    except:
                        print("[Error]. Problem while evaluation expression.")
           
I hope you find this issue report useful, this is my first time so I don't know if it's the right way or not.

akosthekiss, 2017-10-11:
Duplicate of #1900

akosthekiss, 2018-05-10:
Already marked as duplicate but forgot to close. Now closing.

closed 2018-05-10 by akosthekiss