jerryscript-project/jerryscript #2040: native-backed functions have no "prototype" property

gabrielschulhof, 2017-10-10:
The "prototype" property of a function created with `jerry_create_external_function ()` is undefined. This means that one cannot construct classes and then expose them to JS.

We have `jerry_value_is_constructor ()` which allows one to determine if a value can be used as a constructor, and we have `jerry_construct_object ()` which allows one to instantiate a new object using an existing constructor, but we don't have `jerry_create_constructor (jerry_external_handler_t native_function_p)`, which would allow one to create a new constructor and then add properties to its prototype and static ones to the constructor itself.

My motivation is to implement N-API's [`napi_define_class ()`](https://docs.nodejs.org/dist/latest-v8.x/docs/api/n-api.html#n_api_napi_define_class).

Illustrated with a unit test:
```c
#include <string.h>
#include "test-common.h"
#include "jerryscript.h"

static jerry_value_t
native_function (const jerry_value_t function_obj,
                 const jerry_value_t this_val,
                 const jerry_value_t args_p[],
                 const jerry_value_t args_count)
{
  (void) function_obj;
  (void) this_val;
  (void) args_p;
  (void) args_count;
  return jerry_create_undefined();
}

int
main (int argc, char **argv)
{
  (void) argc;
  (void) argv;

  jerry_init (JERRY_INIT_EMPTY);

  jerry_value_t js_native_function = jerry_create_external_function (native_function);
  jerry_value_t global = jerry_get_global_object ();
  jerry_value_t name = jerry_create_string ((jerry_char_t *) "native_function");
  jerry_value_t set_result = jerry_set_property (global, name, js_native_function);
  TEST_ASSERT (!jerry_value_has_error_flag (set_result));

  char *js_code = "typeof native_function.prototype";
  jerry_value_t parse_result = jerry_parse ((jerry_char_t *) js_code,
                                            strlen (js_code),
                                            true);
  TEST_ASSERT (!jerry_value_has_error_flag (parse_result));

  jerry_value_t eval_result = jerry_run (parse_result);
  TEST_ASSERT (!jerry_value_has_error_flag (eval_result));
  TEST_ASSERT (jerry_value_is_string (eval_result));

  jerry_size_t result_size = jerry_get_string_size (eval_result);
  jerry_char_t result_string[result_size + 1];
  jerry_string_to_char_buffer (eval_result, result_string, result_size);
  result_string[result_size] = 0;
  TEST_ASSERT (!strcmp ((char *) result_string, "object"));

  jerry_release_value (eval_result);
  jerry_release_value (parse_result);
  jerry_release_value (set_result);
  jerry_release_value (name);
  jerry_release_value (global);
  jerry_release_value (js_native_function);
  jerry_cleanup();
}
```

LaszloLango, 2017-10-10:
@gabrielschulhof what is the problem with [`jerry_set_prototype`](http://jerryscript.net/api-reference/#jerry_set_prototype)? You can set the prototype with the current API.

gabrielschulhof, 2017-10-10:
@LaszloLango I changed the test to use that, but `jerry_set_prototype ()` doesn't seem to do what I want. I made a [gist](https://gist.github.com/gabrielschulhof/a7a4de122349e67fc79c9ed533ad4730) which can be dropped into tests/unit-ext.

Ideally, I'd want the test to output
```
typeof native_function.prototype -> object
typeof native_function.__proto__ -> <don't care>
typeof (new native_function()).some_property -> string
```
Currently all of the above evaluate to `undefined`.

gabrielschulhof, 2017-10-10:
I basically need to be able to do this from C:
```js
var MyClass = function() {
  if (!(this instanceof MyClass)) {
    return new MyClass();
  }
  this.something = "something else";
};

MyClass.prototype.some_property = "some value";
MyClass.prototype.some_method = function() {
  return "some method";
};
MyClass.someStaticMethod = function() {
  return "some static method";
};
```

gabrielschulhof, 2017-10-11:
Actually, attempting to call `jerry_set_prototype ()` on an external function object results in a `TypeError`.

gabrielschulhof, 2017-10-11:
OK, I'm confused. I changed my [gist](https://gist.github.com/gabrielschulhof/a7a4de122349e67fc79c9ed533ad4730) like [so](https://gist.github.com/gabrielschulhof/a7a4de122349e67fc79c9ed533ad4730/revisions#diff-084b957617be529437d95b01a9c73033), and setting the prototype does not result in an error, whereas making the change
```diff
diff --git a/jerry-main/main-unix.c b/jerry-main/main-unix.c
index 4d2b4167..0b635b20 100644
--- a/jerry-main/main-unix.c
+++ b/jerry-main/main-unix.c
@@ -264,6 +264,18 @@ register_js_function (const char *name_p, /**< name of the function */
     jerry_port_log (JERRY_LOG_LEVEL_WARNING, "Warning: failed to register '%s' method.", name_p);
     print_unhandled_exception (result_val);
   }
+  else
+  {
+    jerry_value_t proto = jerry_create_object ();
+    jerry_value_t set_result = jerry_set_prototype (result_val, proto);
+    if (jerry_value_has_error_flag (set_result))
+    {
+      jerry_port_log (JERRY_LOG_LEVEL_WARNING, "Warning: failed to set prototype for '%s' method: ", name_p);
+      print_unhandled_exception (set_result);
+    }
+    jerry_release_value (set_result);
+    jerry_release_value (proto);
+  }
 
   jerry_release_value (result_val);
 } /* register_js_function */
```
Does result in a `TypeError`.

gabrielschulhof, 2017-10-11:
OK, NM. I think it's because `jerryx_handler_register_global ()` returns the result of setting the value on the global, not the function itself.

akosthekiss, 2017-10-11:
@gabrielschulhof Seems to be a reasonable change to the semantics of `jerryx_handler_register_global` to make it return the created external function. Might be more useful to the caller. Such a change shouldn't break existing code since success/failure is already determined by checking the error flag of the result, and the result is not used in other ways, but must always be released. What do you think?

LaszloLango, 2018-02-16:
I'm closing this due to the long time inactivity. Feel free to reopen if needed.

closed 2018-02-16 by LaszloLango