jerryscript-project/jerryscript #2041: System allocator is not supported for x86_64

tadziopazur, 2017-10-10:
Setting FEATURE_SYSTEM_ALLOCATOR for an x86_64 host build does not work.
jerry/jerry-core/jmem/jmem-allocator.c:69..72
>#if defined (ECMA_VALUE_CAN_STORE_UINTPTR_VALUE_DIRECTLY) && defined (JERRY_CPOINTER_32_BIT)
>  JERRY_ASSERT (((jmem_cpointer_t) uint_ptr) == uint_ptr);
>#else /* !ECMA_VALUE_CAN_STORE_UINTPTR_VALUE_DIRECTLY || !JERRY_CPOINTER_32_BIT */
>  const uintptr_t heap_start = **(uintptr_t) &JERRY_HEAP_CONTEXT (first);**

The JERRY_HEAP_CONTEXT is used unless JERRY_CPOINTER_32_BIT has been set (OK) and ECMA_VALUE_CAN_STORE_UINTPTR_VALUE_DIRECTLY has been computed to true, which is not the case on LP64 programming model employed by GCC for x86_64.

The build process bails out with the following error:

>/opt/iotjs/deps/jerry/jerry-core/jmem/jmem-allocator.c: In function ‘jmem_compress_pointer’:
>/opt/iotjs/deps/jerry/jerry-core/jmem/jmem-allocator.c:72:45: error: implicit declaration of function >‘JERRY_HEAP_CONTEXT’ [-Werror=implicit-function-declaration]
>   const uintptr_t heap_start = (uintptr_t) &JERRY_HEAP_CONTEXT (first);

I care for this setting, since I'd like to delve into valgrind verification of jerry/iotjs integration. With internal allocator that is opaque for valgrind, I'd need to annotate allocator functions for valgrind, which might be useful down the road, but just gets in the way now.

akosthekiss, 2017-10-10:
Background:
- The engine has two concepts for pointers. There are the C pointers, which are used in computations, and there is an internal representation which can be either 16 or 32 bits, which is used for storing pointers in memory.
- If C pointers are longer than the internal representation, a compression happens. There must be a designated base address, from which all memory blocks are max 2^(INTERNAL_PTR_SIZE+ALIGNMENT) bytes away. The compression is to calculate the pointer's offset from the base and then shift it right with the alignment.
- The engine's own allocator is working in an array with a fixed start address, which acts as the base address for compression (if it's necessary because of pointer representation differences).

Unfortunately, the above mean that on 64 bit systems compression must happen (because 64 bit internal pointer representation is not supported), and therefore the system allocator cannot be used (because it cannot be forced to allocate memory in a predefined given range, the start of which could act as the base address of compression).

However, have you tried FEATURE_VALGRIND that adds Valgrind support to the engine's own allocator?

zherczeg, 2017-10-11:
You can also try `--compile-flag='-m32' --cpointer-32bit=on --system-allocator=on --jerry-libc=off`

akosthekiss, 2017-10-17:
@tadziopazur Did any of the answers provide you with a working solution?

tadziopazur, 2017-10-20:
Mostly yes, thank you.

Recently we detected some issues in iotjs for TizenRT, and some of those issues look unrelated to the OS itself, yet do not show up in the valgrind linux testing, which is alarming.

We'll do a follow-up on them once we get ourselves out the woods (release deadline cometh).

closed 2017-10-20 by tadziopazur