jerryscript-project/jerryscript #2053: Math.random() generates same results

tosyu, 2017-10-24:
Because `srand()` is not used on initialization. Without it, the `Math.random()` function generates the same results every time (not every call, but every time the environment runs).

This works on other JS engines like V8 (nodejs). IMHO we should look to achieve same results to be consistent with JS platform

akosthekiss, 2017-10-24:
@tosyu I'm not sure about the reference to other JS engines, because initialising the libc random generator should be the responsibility of the application, not the library (the JS engine). That being said, however, it's true that `srand` is not called anywhere, not even in jerry-main. Will you be submitting a patch to fix that?

zherczeg, 2017-10-24:
do we have srand in jerry libc?

akosthekiss, 2017-10-24:
Note: interestingly, tests/unit-core/test-common.h calls `srand`.

@zherczeg yes, we do.

tosyu, 2017-10-24:
@akosthekiss I did not find anything in the documentation about initializing random number generator with `srand()`, so for an application developer this would require to make assumptions about `Math.random()` implementation, which in my opinion is a "no-no". Either it should be fixed in the code or it should be written in the documentation that `srand()` needs to be called

akosthekiss, 2017-10-24:
@tosyu I see no way for fixing this issue in code (as code of the engine) because library should not (re-)initialise the RNG (might have already been done by the application, and so the lib could cause conflicts). It can and should be fixed in the code of the jerry-main application(s) though. I also agree that it could be documented that `Math.random` is based on libc's `rand`, which would imply the need for initialisation by `srand` (and that could also be written explicitly). (No idea though where to put that note, as we have no dedicated documentation for the JS objects/functions as we assume that everything is documented in the standard. Perhaps the documentation of `jerry_init` in the API docs could have some similar notes. I'm also open to suggestions.)

tosyu, 2017-10-24:
@akosthekiss IMHO `docs/03.API-EXAMPLE.md` would be sufficient to put the information about `srand()`; as an example

akosthekiss, 2017-10-24:
@tosyu That could work, too. Are you up for submitting a PR for this or shall we put that on our todo list?

tosyu, 2017-10-25:
@akosthekiss please put it on the todo list for now

closed 2017-11-03 by LaszloLango