jerryscript-project/jerryscript #2062: How To Force Garbage Collection

natewatson999, 2017-10-27:
Is there a Javascript interface to force Garbage Collection to happen in Jerryscript? I know there's a way to do it from the C perspective, but I can't find a way to do it in JS.

Other JS environments have ways to force garbage collection in JS. Opera's window.opera.collect() and Node's global.gc() are two examples. Does this concept exist in Jerryscript?

LaszloLango, 2017-10-27:
@natewatson999, yes, http://jerryscript.net/api-reference/#jerry_gc

LaszloLango, 2017-10-27:
@natewatson999 if you would like to call it from JavaScript, than you need this: http://jerryscript.net/ext-reference-handler/#jerryx_handler_gc

galpeter, 2017-10-27:
By default the jerry library does not expose the gc api to the JS side, you'll need to expose the C gc API method by registering it to the JS context (as described by @LaszloLango) in your application.

akosthekiss, 2017-10-28:
@natewatson999 Did any of the above help/work for you? Please close the issue if your question has been answered.

natewatson999, 2017-10-29:
I am testing it out now. It may take a little while to be sure because I'm new to Jerryscript & I'm having to figure this out for something that is a bit of an edge case. I should be able to close this or report if it works by tomorrow though.

natewatson999, 2017-10-29:
I think I have it working. Thank you all.

closed 2017-10-29 by natewatson999