jerryscript-project/jerryscript #2065: Assertion 'object_p->type_flags_refs >= ECMA_OBJECT_REF_ONE' failed in ecma_deref_object

renatahodovan, 2017-10-29:
###### Jerry version:

```
Checked revision: d9cc3fd
Build command: ./tools/build.py --clean --debug  --profile=es2015-subset
```
###### OS:

```
Ubuntu 16.04.3 LTS
```
###### Test case:

```javascript
var a = Promise.prototype;
for (var i = 0; i < 200; i++)
{
    Promise.race([a]).then();
}
```

###### Backtrace:

```
ICE: Assertion 'object_p->type_flags_refs >= ECMA_OBJECT_REF_ONE' failed at jerryscript/jerry-core/ecma/base/ecma-gc.c(ecma_deref_object):143.
Error: ERR_FAILED_INTERNAL_ASSERTION
bt
#0  0x000000000059e248 in raise ()
#1  0x000000000059e45a in abort ()
#2  0x000000000057e4bd in jerry_port_fatal (code=25351) at jerryscript/jerry-port/default/default-fatal.c:71
#3  0x00000000004faeef in jerry_fatal (code=<optimized out>) at jerryscript/jerry-core/jrt/jrt-fatals.c:60
#4  0x00000000004fafe0 in jerry_assert_fail (assertion=0x6 <error: Cannot access memory at address 0x6>, file=0x59e248 <raise+56> "H=", function=0xb01880 "\200\030\260", line=<optimized out>) at jerryscript/jerry-core/jrt/jrt-fatals.c:84
#5  0x0000000000418742 in ecma_deref_object (object_p=<optimized out>) at jerryscript/jerry-core/ecma/base/ecma-gc.c:143
#6  0x00000000004dd48d in ecma_process_promise_reaction_job (obj_p=0xa87920 <jerry_global_heap+31728>) at jerryscript/jerry-core/ecma/operations/ecma-jobqueue.c:210
#7  0x00000000004ddcf9 in ecma_process_all_enqueued_jobs () at jerryscript/jerry-core/ecma/operations/ecma-jobqueue.c:344
#8  0x0000000000402f85 in main (argc=<optimized out>, argv=0x7fffffffd188) at jerryscript/jerry-main/main-unix.c:900
```

<sup>Found by [Fuzzinator](http://fuzzinator.readthedocs.io/). </sup>

jiangzidong, 2017-12-12:
Fixed by #2115  . This issue and #2107 share the same root-cause.

jiangzidong, 2017-12-12:
Sorry that I missed this issue, and just found it right now...

closed 2017-12-12 by jiangzidong