jerryscript-project/jerryscript #2077: arm-none-eabi-ld: region `sram' overflowed by 16492 bytes

Zakaria-Ben, 2017-11-08:
Hi Guys,
I want to build Jerryscript for NuttX on a NUCLEO-F401RE board.
I followed the instructions exactly in the https://github.com/jerryscript-project/jerryscript/tree/master/targets/nuttx-stm32f4 except for step 3, I used ./configure.sh nucleo-f4x1re/f401-nsh.
And when I try to build JerryScript for NuttX (Step 4) using :
cd nuttx/
make
I got this error : 

make[2]: Leaving directory '/home/ubuntu/jerry-nuttx/nuttx/configs/nucleo-f4x1re/src'
LD: nuttx
arm-none-eabi-ld: /home/ubuntu/jerry-nuttx/nuttx/nuttx section `.bss' will not fit in region `sram'
arm-none-eabi-ld: region `sram' overflowed by 16492 bytes
Makefile:203: recipe for target 'nuttx' failed
make[1]: *** [nuttx] Error 1
make[1]: Leaving directory '/home/ubuntu/jerry-nuttx/nuttx/arch/arm/src'
Makefile.unix:416: recipe for target 'pass2' failed
make: *** [pass2] Error 2

I'm really struggling with this error

galpeter, 2017-11-09:
I think the problem is that the  STM32F4-Discovery board has more memory than the NUCLEO and the jerry heap fits into the discovery but not on the nucleo. You can decrease the heap size in the makefile:
https://github.com/jerryscript-project/jerryscript/blob/master/targets/nuttx-stm32f4/Makefile#L21

Zakaria-Ben, 2017-11-09:
Thank you so much, it works, but in fact I changed both the STACKSIZE and the HEAPSIZE. but,
what is the consequence of that ???

Zakaria-Ben, 2017-11-09:
Sorry, But it doesn't work. I used the make clean and then I forgot to do the configuration from the menuconfig in order to add JerryScript interpreter.

Zakaria-Ben, 2017-11-09:
galpeter, it's fine :D . Thanks for your help.

galpeter, 2017-11-14:
In simple terms, the HEAPSIZE defines how much live objects/data can be in the memory in a given time and also makes the garbage collector run less in a given time frame (as the system has more heap no need to remove old objects that often). The JerryScript heap is pre-allocated in the bss secion which will placed into the sram. That's why if you give bigger heap it can go out of the sram size.

The STACKSIZE is afaik a nuttx config to specify each application total stack size, that is how much variables/nested function calls can be stored on the stack.

Is there anything else? Or can we close this issue?

Zakaria-Ben, 2017-11-14:
That's fine. Thank you for your help.

galpeter, 2017-11-14:
You're welcome, closing issue.

closed 2017-11-14 by galpeter