jerryscript-project/jerryscript #2080: Changes from #2042 break semantic API compatibility

martijnthe, 2017-11-09:
Before PR #2042, `jerry_value_clear_error_flag` and `jerry_value_set_error_flag` would take a `jerry_value_t *` and set a flag onto it. After each of these calls return, the `jerry_value_t` that the pointer points to would still be a reference to the same JS thing as before.

After PR #2042, after the `jerry_value_set_error_flag` returns, the `jerry_value_t` is updated to reference a *different* value (the error "box"). Same with `jerry_value_clear_error_flag`, it expects to take an `jerry_value_t *` pointing to such an error "box" and after the call returns, the `jerry_value_t` will contain the unboxed value.

Even though this change is source and binary compatible, semantically it's not compatible (and thus breaking the API, IMHO).

One of the more serious implications is that the client code needs to be aware of the "swapping" of the `jerry_value_t`:

Contrived example:
```
jerry_value_t sth = ...;

sth = jerry_acquire_value(sth);  // this worked fine before #2042
jerry_value_set_error_flag(&sth);

// Whoops! After PR #2042 the boxed value is not acquired...
return sth;
```

Unless we really need to free up the error flag bit in the `jerry_value_t` *now*, I'd revert #2042, I'm afraid it will break people's projects in unexpected ways. At the very least, the docs should be updated – this change is more than a transparent implementation detail.

If the freed up bit is actually really needed now, perhaps create a new API with a name that better matches what happens under the hood, mark the old one deprecated. The old one can call the new implementation, but at least this way there's a mechanism to get people's attention after pulling from upstream.

cc @zherczeg @akosthekiss

martijnthe, 2017-11-09:
cc @jiangzidong

jiangzidong, 2017-11-10:
I can provide another example which we met in our real project.

the code is like
```c
{
  jerry_value_t rv = get_a_jerry_value_from_some_api(...);

  // in the js_check_and_log_if_error, we check if rv is error, 
  // if so, call jerry_value_clear_error_flag, and print the error log 
  if (js_check_and_log_if_error(rv)){
    // error handle
  } else {
     // normal handle
  }

  jerry_release_value(rv);
}
```

After we apply the patch #2042 , the above code can't work, because the `jerry_value_clear_error_flag` will totally change the value of rv, but only inside the  `js_check_and_log_if_error`, the rv in outside still same. so the `jerry_release_value(rv);` will not release the error's real value. Besides it may cause stange error, because the error_box is already released in `jerry_value_clear_error_flag`

martijnthe, 2017-11-10:
To provide a bit more context, the reason we are calling `jerry_value_clear_error_flag` in the first place, is because we want to call `jerry_value_to_string(rv)` to get the error message and print it. But `jerry_value_to_string` doesn't work unless the error "flag" is removed.

If we can avoid using `jerry_value_clear_error_flag` and `jerry_value_set_error_flag` in the first place, I think we can avoid falling into the traps that #2042 has created.

One way would be to have an API to get the non-error value associated with the error "box" value.
So basically, add a new API to `jerry.c`:

```
jerry_value_t jerry_get_value_without_error_flag(jerry_value_t value) {
  return jerry_acquire_value(jerry_get_arg_value(value));
}
```

@akosthekiss @zherczeg what do you think about this idea?

zherczeg, 2017-11-14:
Hi guys!

You described a way of misusing the API. The point of passing the pointer to set/clear flag was the liberty of doing anything with the value, and you must use the modified value afterwards (your original reference is dropped, a new reference is created). Using the original dropped reference can lead to crashes.

zherczeg, 2017-11-14:
(I have to admit misusing the api was not have a fatal effect before, but that does not mean the misusing is allowed).

martijnthe, 2017-11-14:
@zherczeg ok, thanks for clarifying.
What do you think about the API idea in my [last comment]?

[last comment]: https://github.com/jerryscript-project/jerryscript/issues/2080#issuecomment-343445688

zherczeg, 2017-11-14:
I think we should specify its purpose more clearly. Something like:
 - if value has error flag, remove the flag and returns a new reference to the value
 - if value has no error flag, it just creates a new reference
Hence it never drops the current reference.

martijnthe, 2017-11-14:
@jiangzidong could you take a stab at creating a PR for that ^^^

jiangzidong, 2017-11-14:
Ok @martijnthe

jiangzidong, 2017-11-14:
> if value has error flag, remove the flag and returns a new reference to the value

For error value, we just need to get its reference value, and dont need to remove its error_flag. The API is just get the "real" value. I think @martijnthe 's proposal is enough for this issue.

closed 2018-02-16 by LaszloLango