jerryscript-project/jerryscript #2085: riot-stm32f4 fails to build

mfrey, 2017-11-10:
Hi,

I want to build jerryscript for RIOT. If I try to build the source via
```
make -f ./targets/riot-stm32f4/Makefile.riot
```
I get the following errors
```c
[  0%] Building C object jerry-core/CMakeFiles/jerry-core.dir/ecma/operations/ecma-regexp-object.c.obj
/home/frey/Desktop/Projects/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c: In function 're_match_regexp':
/home/frey/Desktop/Projects/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:747:7: error: this statement may fall through [-Werror=implicit-fallthrough=]
       {
       ^
/home/frey/Desktop/Projects/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:801:7: note: here
       case RE_OP_CAPTURE_GROUP_START:
       ^~~~
/home/frey/Desktop/Projects/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:933:14: error: this statement may fall through [-Werror=implicit-fallthrough=]
         bc_p = old_bc_p;
         ~~~~~^~~~~~~~~~
/home/frey/Desktop/Projects/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:938:7: note: here
       case RE_OP_CAPTURE_GREEDY_GROUP_END:
       ^~~~
cc1: some warnings being treated as errors
```
Compiling jerryscript via
```
python tools/build.py
```
works fine. Am I missing something?

Thanks in advance!

mfrey, 2017-11-10:
Added the full [output.log](https://github.com/jerryscript-project/jerryscript/files/1461483/output.log) to the issue

LaszloLango, 2017-11-10:
@mfrey hi. The RIOT build works for me. I have no idea why you got such errors. What are your GCC versions (x86 and arm-noneabi)?

```
$ gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.5) 5.4.0 20160609
```
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)
```

mfrey, 2017-11-10:
@LaszloLango Hi, my gcc version is
```
gcc (GCC) 7.2.0
```
and  arm-gcc is
```
arm-none-eabi-gcc (Arch Repository) 7.2.0
```

mfrey, 2017-11-10:
Apparently gcc 7 has added a default fallthrough warning behaviour. It also introduces a mechanism for so called "fallthrough indicators" where you can pass regular expressions to it. 

See also the [documentation](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html) for more details.

LaszloLango, 2017-11-10:
Issues and PRs that might be related: #1878, #1964, #1965

LaszloLango, 2017-11-10:
Try verbose build and check whether `-Wno-implicit-fallthrough` is added or not.

`VERBOSE=1 make -f ./targets/riot-stm32f4/Makefile.riot`

mfrey, 2017-11-10:
Unfortunately, it's not added.
[warnings.log](https://github.com/jerryscript-project/jerryscript/files/1461619/warnings.log)

LaszloLango, 2017-11-10:
@mfrey you must have some problem with this conditions: https://github.com/jerryscript-project/jerryscript/blob/master/CMakeLists.txt#L236

For a workaround, you can comment it out to always add the necessary compiler flag.

dbatyai, 2017-11-10:
The problem was that CMake did not correctly set the compiler version. I've submitted a fix (#2088), @mfrey could you please check if it works?

mfrey, 2017-11-10:
@dbatyai Works for me! 

@LaszloLango I've set it by hand in the Makefile.riot file (and that works as well)! 

Thank you for your help!

mfrey, 2018-02-05:
Closed this issue since it is solved! Thanks!

closed 2018-02-05 by mfrey