jerryscript-project/jerryscript #2093: Zephyr target fails to build with latest Zephyr cmake changes

jimmy-huang, 2017-11-13:
After updating Zephyr to master, JerryScript will no longer build, since Zephyr now has changed to build with cmake:

$ make -f targets/zephyr/Makefile.zephyr 
targets/zephyr/Makefile.zephyr:47: build/arduino_101/zephyr/Makefile.export: No such file or directory
make -f ./targets/zephyr/Makefile O="build/arduino_101/zephyr" V= USER_LIBS="jerry-core jerry-ext" USER_LIB_INCLUDE_DIR="-L /home/jimmy/projects/zephyrjs/zephyr.js/deps/jerryscript/build/arduino_101/obj-arduino_101/lib" TARGET_ZEPHYR=./targets/zephyr initconfig
make[1]: Entering directory '/home/jimmy/projects/zephyrjs/zephyr.js/deps/jerryscript'
targets/zephyr/Makefile:60: /home/jimmy/projects/zephyrjs/zephyr.js/deps/zephyr/Makefile.inc: No such file or directory
make[1]: *** No rule to make target '/home/jimmy/projects/zephyrjs/zephyr.js/deps/zephyr/Makefile.inc'.  Stop.
make[1]: Leaving directory '/home/jimmy/projects/zephyrjs/zephyr.js/deps/jerryscript'
targets/zephyr/Makefile.zephyr:103: recipe for target 'build/arduino_101/zephyr/include/config/auto.conf' failed
make: *** [build/arduino_101/zephyr/include/config/auto.conf] Error 2

jimmy-huang, 2018-04-09:
Closing this as it is resolved.

closed 2018-04-09 by jimmy-huang

xiangchaosheng, 2018-09-21:
Hello, jimmy. how do you solved this problem? can you share it with us? thanks~