jerryscript-project/jerryscript #2094: dynamic-stack-buffer-overflow in jerry_value_is_syntax_error

renatahodovan, 2017-11-14:
###### Jerry version:

```
Checked revision: 6b1ed6e0
Build command: CC=clang ./tools/build.py --clean --debug --compile-flag=-fsanitize=address --compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer --compile-flag=-fno-common --jerry-libc=off --static-link=off --lto=off --error-messages=on
```
###### OS:

```
Ubuntu 17.10
macOS High Sierra (10.13.1)
```
###### Test case:

It seems that jerry produces the same failure with any kind of syntax errors when compiling it with clang. E.g.:

```javascript
function a() {
```

###### Backtrace:

```
=================================================================
==1467==ERROR: AddressSanitizer: dynamic-stack-buffer-overflow on address 0xff934fcb at pc 0x080abfde bp 0xff934f88 sp 0xff934b60
READ of size 12 at 0xff934fcb thread T0
    #0 0x80abfdd in __interceptor_strcmp.part.32 crtstuff.c:?
    #1 0x816f8e3 in jerry_value_is_syntax_error jerryscript/jerry-main/main-unix.c:122
    #2 0x816e8e9 in print_unhandled_exception jerryscript/jerry-main/main-unix.c:154 (discriminator 1)
    #3 0x816d99f in main jerryscript/jerry-main/main-unix.c:790
    #4 0xf7be3985 in __libc_start_main ??:?
    #5 0x806e177 in _start ??:?

Address 0xff934fcb is located in stack of thread T0
SUMMARY: AddressSanitizer: dynamic-stack-buffer-overflow crtstuff.c:? in __interceptor_strcmp.part.32
Shadow bytes around the buggy address:
  0x3ff269a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x3ff269b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x3ff269c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x3ff269d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x3ff269e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x3ff269f0: 00 00 00 00 ca ca ca ca 00[03]cb cb cb cb cb cb
  0x3ff26a00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x3ff26a10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x3ff26a20: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x3ff26a30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x3ff26a40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==1467==ABORTING

```

<sup>Found by [Fuzzinator](http://fuzzinator.readthedocs.io/) with [grammarinator](https://github.com/renatahodovan/grammarinator). </sup>

rerobika, 2017-11-14:
I'll check it.

rerobika, 2017-11-20:
#2095 is landed. This issue can be closed.

closed 2017-11-20 by akosthekiss