jerryscript-project/jerryscript #2098: JerryScript issue on Nuttx

Zakaria-Ben, 2017-11-14:
Hi Guys,
I'm want to use JerryScript on Nuttx. I've created a RAMDISK using mkrd, then I used echo in order to create a file with a .js extension, but when I run jerry /mnt/fs/test.js, I got a message saying : 
Error: cannot open file: test.js                                                
Source file load error 

the test.js file is just a console.log .

LaszloLango, 2017-11-14:
cc: @rtakacs

galpeter, 2017-11-14:
Could you try the following?
1) change to the /mnt/fs/ directory
2) run: jerry test.js

Other questions:
  * Does the  cat /mnt/fs/test.js  prints out the file contents correctly?
  * Did you registered a  'console.log' method into the JerryScript nuttx ap? If not that then your script will throw a reference error. You should use the 'print' method instead as it is in included in the target.
  * Was the the CCM memory enabled during Nuttx build?

Zakaria-Ben, 2017-11-14:
@galpeter Thank you so much, it works perfectly with the print. but how can I register the 'console.log' ?
I want to ask you another question, because I just start using Nuttx and jerryScript, I want to implement Wamp protocol using Autobahn : https://github.com/crossbario/autobahn-js. I'm using a Nucleo32F4 board, what do you think about the choice of Nuttx with Jerryscript ?

martijnthe, 2017-11-14:
> but how can I register the 'console.log' ?

Some examples of other projects and how they implement `console`:

https://github.com/01org/zephyr.js/blob/62fd1b952450c788920ea45d76f068f06ebad983/src/zjs_console.c#L246
https://github.com/Samsung/iotjs/blob/932980a22abf19261ad8f4a9299a06093b268984/src/modules/iotjs_module_console.c#L50

I guess you could also implement it in JS by wrapping `print`, i.e.:
```
console = {
  log: print,
  error: print,
  warn: print,
};
```

Zakaria-Ben, 2017-11-15:
Many thanks @martijnthe

zherczeg, 2018-01-29:
I am closing this bug. Please reopen if needed.

Txiaozhe, 2019-12-02:
Why do not you support built-in console in jerryscript?

akosthekiss, 2019-12-02:
@Txiaozhe JerryScript is a JavaScript engine that is implementing built-ins that are described in the JavaScript (i.e., EcmaScript) standard. AFAIK `console` is not such a built-in. So, implementing it is the task of the embedder of the engine, e.g., IoT.js.

Txiaozhe, 2019-12-02:
Yeah! Thanks:)