jerryscript-project/jerryscript #2104: Assertion '!ecma_is_value_direct (x) || ecma_is_value_undefined (x) || ecma_is_value_null (x) || ecma_is_value_boolean (x) || ecma_is_value_integer_number (x)' failed in ecma_op_strict_equality_compare

renatahodovan, 2017-11-16:
###### Jerry version:

```
Checked revision: 7692aa9d
Build command: ./tools/build.py --clean --debug --profile=es2015-subset
```
###### OS:

```
Ubuntu 17.10
```
###### Test case:

```javascript
var value;
Object.defineProperty (Array.prototype, 'then', {get: function() { throw$ } })
promise = Promise.all ([])
promise.then (
    function() {},
    function(val) {
        if (val !== value) ;
    })

```

###### Backtrace:

```
ICE: Assertion '!ecma_is_value_direct (x) || ecma_is_value_undefined (x) || ecma_is_value_null (x) || ecma_is_value_boolean (x) || ecma_is_value_integer_number (x)' failed at jerryscript/jerry-core/ecma/operations/ecma-comparison.c(ecma_op_strict_equality_compare):205.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0xf7fd5db9 in __kernel_vsyscall ()
(gdb) bt
#0  0xf7fd5db9 in __kernel_vsyscall ()
#1  0xf7816352 in raise () from /lib32/libc.so.6
#2  0xf7817a51 in abort () from /lib32/libc.so.6
#3  0x56574406 in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-port/default/default-fatal.c:71
#4  0x565e331b in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.c:58
#5  0x565e335c in jerry_assert_fail (
    assertion=0x56682420 "!ecma_is_value_direct (x) || ecma_is_value_undefined (x) || ecma_is_value_null (x) || ecma_is_value_boolean (x) || ecma_is_value_integer_number (x)", file=0x56682340 "jerryscript/jerry-core/ecma/operations/ecma-comparison.c", 
    function=0x56675740 <__func__.2776.lto_priv.213> "ecma_op_strict_equality_compare", line=205)
    at jerryscript/jerry-core/jrt/jrt-fatals.c:82
#6  0x56581626 in ecma_op_strict_equality_compare (x=24, y=72) at jerryscript/jerry-core/ecma/operations/ecma-comparison.c:201
#7  0x56604dc6 in vm_loop (frame_ctx_p=0xffffcb10) at jerryscript/jerry-core/vm/vm.c:1922
#8  0x5660a0bc in vm_execute (frame_ctx_p=0xffffcb10, arg_p=0xf5d000d4, arg_list_len=1) at jerryscript/jerry-core/vm/vm.c:2795
#9  0x5660a960 in vm_run (bytecode_header_p=0xf5f009d0, this_binding_value=4126149459, lex_env_p=0xf5d007b0, is_eval_code=false, 
    arg_list_p=0xf5d000d4, arg_list_len=1) at jerryscript/jerry-core/vm/vm.c:2884
#10 0x566141f5 in ecma_op_function_call (func_obj_p=0xf5f00250, this_arg_value=72, arguments_list_p=0xf5d000d4, arguments_list_len=1)
    at jerryscript/jerry-core/ecma/operations/ecma-function-object.c:406
#11 0x56616526 in ecma_process_promise_reaction_job (obj_p=0xf5d000d0)
    at jerryscript/jerry-core/ecma/operations/ecma-jobqueue.c:170
#12 0x566172d4 in ecma_process_all_enqueued_jobs () at jerryscript/jerry-core/ecma/operations/ecma-jobqueue.c:344
#13 0x5664e892 in jerry_run_all_enqueued_jobs () at jerryscript/jerry-core/api/jerry.c:596
#14 0x5664b773 in main (argc=3, argv=0xffffd094) at jerryscript/jerry-main/main-unix.c:797
```

<sup>Found by [Fuzzinator](http://fuzzinator.readthedocs.io/) with  [grammarinator](https://github.com/renatahodovan/grammarinator). </sup>

fbmrk, 2017-11-22:
Can't reproduce.
Maybe fixed in #2116

renatahodovan, 2018-01-09:
@fbmrk Me neither. Can be closed.

closed 2018-01-09 by renatahodovan