jerryscript-project/jerryscript #2108: Assertion 'lit_is_valid_cesu8_string (string_p, string_size)' failed in ecma_new_ecma_string_from_utf8

renatahodovan, 2017-11-16:
###### Jerry version:

```
Checked revision: 7692aa9d
Build command: ./tools/build.py --clean --debug
```
###### OS:

```
Ubuntu 17.10
```
###### Test case:

```javascript
print(new Number(8e-9).toFixed(20));
```

###### Backtrace:

```
ICE: Assertion 'lit_is_valid_cesu8_string (string_p, string_size)' failed at jerryscript/jerry-core/ecma/base/ecma-helpers-string.c(ecma_new_ecma_string_from_utf8):154.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0xf7fd5db9 in __kernel_vsyscall ()
(gdb) bt
#0  0xf7fd5db9 in __kernel_vsyscall ()
#1  0xf7816352 in raise () from /lib32/libc.so.6
#2  0xf7817a51 in abort () from /lib32/libc.so.6
#3  0x565743e6 in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION)
    at jerryscript/jerry-port/default/default-fatal.c:71
#4  0x565e32fb in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.c:58
#5  0x565e333c in jerry_assert_fail (assertion=0x56697080 "lit_is_valid_cesu8_string (string_p, string_size)", 
    file=0x56696f20 "jerryscript/jerry-core/ecma/base/ecma-helpers-string.c", 
    function=0x56665300 <__func__.3043.lto_priv.642> "ecma_new_ecma_string_from_utf8", line=154)
    at jerryscript/jerry-core/jrt/jrt-fatals.c:82
#6  0x56627f03 in ecma_new_ecma_string_from_utf8 (string_p=0xf5f008e0 "0.000000008", string_size=22)
    at jerryscript/jerry-core/ecma/base/ecma-helpers-string.c:154
#7  0x565b7a3c in ecma_builtin_number_prototype_object_to_fixed (this_arg=4126148979, arg=320)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-number-prototype.c:691
#8  0x565b508a in ecma_builtin_number_prototype_dispatch_routine (builtin_routine_id=61, this_arg_value=4126148979, 
    arguments_list=0xffffcb70, arguments_number=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-number-prototype.inc.h:37
#9  0x56647ed6 in ecma_builtin_dispatch_routine (builtin_object_id=ECMA_BUILTIN_ID_NUMBER_PROTOTYPE, builtin_routine_id=61, 
    this_arg_value=4126148979, arguments_list=0xffffcb70, arguments_number=1)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:91
#10 0x56648652 in ecma_builtin_dispatch_call (obj_p=0xf5f00940, this_arg_value=4126148979, arguments_list_p=0xffffcb70, 
    arguments_list_len=1) at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.c:844
#11 0x56613f96 in ecma_op_function_call (func_obj_p=0xf5f00940, this_arg_value=4126148979, arguments_list_p=0xffffcb70, 
    arguments_list_len=1) at jerryscript/jerry-core/ecma/operations/ecma-function-object.c:343
#12 0x565fd55b in opfunc_call (frame_ctx_p=0xffffcbe0) at jerryscript/jerry-core/vm/vm.c:425
#13 0x5660a0fd in vm_execute (frame_ctx_p=0xffffcbe0, arg_p=0x0, arg_list_len=0) at jerryscript/jerry-core/vm/vm.c:2804
#14 0x5660a940 in vm_run (bytecode_header_p=0xf5300f50, this_binding_value=4126149459, lex_env_p=0xf5d007b0, is_eval_code=false, 
    arg_list_p=0x0, arg_list_len=0) at jerryscript/jerry-core/vm/vm.c:2884
#15 0x565fca81 in vm_run_global (bytecode_p=0xf5300f50) at jerryscript/jerry-core/vm/vm.c:232
#16 0x5664e8bc in jerry_run (func_val=4126149123) at jerryscript/jerry-core/api/jerry.c:558
#17 0x5664b236 in main (argc=3, argv=0xffffd024) at jerryscript/jerry-main/main-unix.c:657
```

<sup>Found by [Fuzzinator](http://fuzzinator.readthedocs.io/) with [grammarinator](https://github.com/renatahodovan/grammarinator). </sup>

DanielBallaSZTE, 2017-11-16:
Can't reproduce on Linux Mint 17.2

I get `0.000000008\u0000\u000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000` as the output.

loki04, 2017-11-16:
It can be reproduced on 
```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 17.10
Release:        17.10
Codename:       artful

$ uname -a
Linux 4.13.0-16-generic #19-Ubuntu SMP Wed Oct 11 18:35:14 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```

DanielBallaSZTE, 2017-11-16:
Hmm, it might be the fact that I'm using gcc 4.8.4

glistening, 2017-11-17:
It is reproduced on my Ubuntu 14.04.3 LTS with gcc 4.9.4.

PR #2112 (2017-11-17): Fixed toFixed method string conversion
This patch fixes #2108. The problem was if the convertible number had less significant fractions digits than the requested, the result was filled with memory junk instead of zeros.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
commit 1dedc1b6 (2017-11-17): Fixed toFixed method string conversion (#2112)

This patch fixes #2108. The problem was if the convertible number had less significant fractions digits than the requested, the result was filled with memory junk instead of zeros.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
closed 2017-11-17 by LaszloLango