jerryscript-project/jerryscript #2118: error: "UINTPTR_MAX" is not defined

wangnuannuan, 2017-11-22:
make -f ./targets/esp8266/Makefile.esp8266

/home/wang/jerryscript/jerry-core/ecma/base/ecma-globals.h:115:5: error: "UINTPTR_MAX" is not defined [-Werror=undef]
 #if UINTPTR_MAX <= UINT32_MAX
     ^
cc1: some warnings being treated as errors
jerry-core/CMakeFiles/jerry-core.dir/build.make:62: recipe for target 'jerry-core/CMakeFiles/jerry-core.dir/__/jerry-all-in.c.obj' failed
make[4]: *** [jerry-core/CMakeFiles/jerry-core.dir/__/jerry-all-in.c.obj] Error 1
make[4]: Leaving directory '/home/wang/jerryscript/build/obj-esp8266'
CMakeFiles/Makefile2:140: recipe for target 'jerry-core/CMakeFiles/jerry-core.dir/all' failed
make[3]: *** [jerry-core/CMakeFiles/jerry-core.dir/all] Error 2
make[3]: Leaving directory '/home/wang/jerryscript/build/obj-esp8266'
CMakeFiles/Makefile2:152: recipe for target 'jerry-core/CMakeFiles/jerry-core.dir/rule' failed
make[2]: *** [jerry-core/CMakeFiles/jerry-core.dir/rule] Error 2
make[2]: Leaving directory '/home/wang/jerryscript/build/obj-esp8266'
Makefile:175: recipe for target 'jerry-core' failed
make[1]: *** [jerry-core] Error 2
make[1]: Leaving directory '/home/wang/jerryscript/build/obj-esp8266'
targets/esp8266/Makefile.esp8266:44: recipe for target 'jerry' failed
make: *** [jerry] Error 2

akosthekiss, 2017-11-22:
@wangnuannuan Please provide info on your build environment (OS type/distribution/version, compiler type/version) and the version of JerryScript you are trying to build (git hash). If you can, try compare your build environment to the system of Travis CI. Build testing of the ESP8266 target has recently been added to the project, and the master branch is green on the CI, so it's especially strange that you are experiencing a build error now.

wangnuannuan, 2017-11-22:
ubuntu16.04  64-bit 
Follow this(https://github.com/jerryscript-project/jerryscript/blob/master/targets/esp8266/docs/ESP-PREREQUISITES.md) page to setup build environment

rerobika, 2017-11-22:
@wangnuannuan Could you tell us what the compiler's type and version that you used? I'm almost completely sure that you're using the latest master but to make sure of it please let us know the version of JerryScript you are trying to build.

wangnuannuan, 2017-11-22:
yes,i am using the latest master

akosthekiss, 2017-11-22:
@wangnuannuan Could you do a verbose build so that we get a bit more information on how the compiler is invoked? (It's usually `make -f ... VERBOSE=1`.)

wangnuannuan, 2017-11-22:
This is all the information that the command line outputs

make -f ./targets/esp8266/Makefile.esp8266 VERBOSE=1
mkdir -p build/obj-esp8266
mkdir -p targets/esp8266/libs
cmake -Bbuild/obj-esp8266 -H./ \
 -DCMAKE_SYSTEM_NAME=MCU \
 -DCMAKE_SYSTEM_PROCESSOR=xtensia-lx106 \
 -DCMAKE_C_COMPILER=xtensa-lx106-elf-gcc \
 -DCMAKE_C_COMPILER_WORKS=TRUE \
 -DENABLE_LTO=OFF \
 -DENABLE_ALL_IN_ONE=ON \
 -DJERRY_LIBC=OFF \
 -DJERRY_CMDLINE=OFF \
 -DEXTERNAL_COMPILE_FLAGS="-D__TARGET_ESP8266 -D__attr_always_inline___= -Wl,-EL -fno-inline-functions -ffunction-sections -fdata-sections -mlongcalls -mtext-section-literals -mno-serialize-volatile" \
 -DMEM_HEAP_SIZE_KB=20
Re-run cmake no build system arguments
System is unknown to cmake, create:
Platform/MCU to use this system, please send your config file to cmake@www.cmake.org so it can be added to cmake
Your CMakeCache.txt file was copied to CopyOfCMakeCache.txt. Please send that file to cmake@www.cmake.org.
-- CMAKE_BUILD_TYPE          MinSizeRel
-- CMAKE_SYSTEM_NAME         MCU
-- CMAKE_SYSTEM_PROCESSOR    xtensia-lx106
-- ENABLE_ALL_IN_ONE         ON
-- ENABLE_LTO                OFF
-- ENABLE_STATIC_LINK        ON
-- ENABLE_STRIP              ON
-- JERRY_CMDLINE             OFF
-- JERRY_CMDLINE_TEST        OFF
-- JERRY_CMDLINE_SNAPSHOT    OFF
-- JERRY_PORT_DEFAULT        ON
-- JERRY_EXT                 ON
-- JERRY_LIBC                OFF
-- JERRY_LIBM                ON
-- UNITTESTS                 OFF
-- DOCTESTS                  OFF
-- FEATURE_INIT_FINI         OFF
-- FEATURE_CPOINTER_32_BIT   OFF
-- FEATURE_DEBUGGER          OFF
-- FEATURE_ERROR_MESSAGES    OFF
-- FEATURE_EXTERNAL_CONTEXT  OFF
-- FEATURE_JS_PARSER         ON
-- FEATURE_MEM_STATS         OFF
-- FEATURE_MEM_STRESS_TEST   OFF
-- FEATURE_PARSER_DUMP       OFF
-- FEATURE_PROFILE           es5.1
-- FEATURE_REGEXP_DUMP       OFF
-- FEATURE_SNAPSHOT_EXEC     OFF
-- FEATURE_SNAPSHOT_SAVE     OFF
-- FEATURE_SYSTEM_ALLOCATOR  OFF
-- FEATURE_VALGRIND          OFF
-- FEATURE_VALGRIND_FREYA    OFF
-- FEATURE_VM_EXEC_STOP      OFF
-- MEM_HEAP_SIZE_KB          20
-- Configuring done
-- Generating done
-- Build files have been written to: /home/wang/jerryscript/build/obj-esp8266
make -Cbuild/obj-esp8266 jerry-core jerry-libm
make[1]: Entering directory '/home/wang/jerryscript/build/obj-esp8266'
/usr/bin/cmake -H/home/wang/jerryscript -B/home/wang/jerryscript/build/obj-esp8266 --check-build-system CMakeFiles/Makefile.cmake 0
make -f CMakeFiles/Makefile2 jerry-core
make[2]: Entering directory '/home/wang/jerryscript/build/obj-esp8266'
/usr/bin/cmake -H/home/wang/jerryscript -B/home/wang/jerryscript/build/obj-esp8266 --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/wang/jerryscript/build/obj-esp8266/CMakeFiles 21
make -f CMakeFiles/Makefile2 jerry-core/CMakeFiles/jerry-core.dir/all
make[3]: Entering directory '/home/wang/jerryscript/build/obj-esp8266'
make -f jerry-libm/CMakeFiles/jerry-libm.dir/build.make jerry-libm/CMakeFiles/jerry-libm.dir/depend
make[4]: Entering directory '/home/wang/jerryscript/build/obj-esp8266'
cd /home/wang/jerryscript/build/obj-esp8266 && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/wang/jerryscript /home/wang/jerryscript/jerry-libm /home/wang/jerryscript/build/obj-esp8266 /home/wang/jerryscript/build/obj-esp8266/jerry-libm /home/wang/jerryscript/build/obj-esp8266/jerry-libm/CMakeFiles/jerry-libm.dir/DependInfo.cmake --color=
make[4]: Leaving directory '/home/wang/jerryscript/build/obj-esp8266'
[... 13 lines cut ...]
make[4]: Entering directory '/home/wang/jerryscript/build/obj-esp8266'
[ 95%] Building C object jerry-core/CMakeFiles/jerry-core.dir/__/jerry-all-in.c.obj
cd /home/wang/jerryscript/build/obj-esp8266/jerry-core && /opt/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc  -DCONFIG_DISABLE_ES2015 -DCONFIG_MEM_HEAP_AREA_SIZE=20480 -DJERRY_NDEBUG -I/home/wang/jerryscript/jerry-core -I/home/wang/jerryscript/jerry-core/api -I/home/wang/jerryscript/jerry-core/debugger -I/home/wang/jerryscript/jerry-core/ecma/base -I/home/wang/jerryscript/jerry-core/ecma/builtin-objects -I/home/wang/jerryscript/jerry-core/ecma/builtin-objects/typedarray -I/home/wang/jerryscript/jerry-core/ecma/operations -I/home/wang/jerryscript/jerry-core/include -I/home/wang/jerryscript/jerry-core/jcontext -I/home/wang/jerryscript/jerry-core/jmem -I/home/wang/jerryscript/jerry-core/jrt -I/home/wang/jerryscript/jerry-core/lit -I/home/wang/jerryscript/jerry-core/parser/js -I/home/wang/jerryscript/jerry-core/parser/regexp -I/home/wang/jerryscript/jerry-core/vm -I/home/wang/jerryscript/jerry-libm/include  -fno-builtin -fno-stack-protector -Wall -Werror=all -Wextra -Werror=extra -Wformat-nonliteral -Werror=format-nonliteral -Winit-self -Werror=init-self -Wconversion -Werror=conversion -Wsign-conversion -Werror=sign-conversion -Wformat-security -Werror=format-security -Wmissing-declarations -Werror=missing-declarations -Wshadow -Werror=shadow -Wstrict-prototypes -Werror=strict-prototypes -Wundef -Werror=undef -Wold-style-definition -Werror=old-style-definition -Wno-stack-protector -Wno-attributes -Wlogical-op -Werror=logical-op -std=c99 -pedantic -D__TARGET_ESP8266 -D__attr_always_inline___= -Wl,-EL -fno-inline-functions -ffunction-sections -fdata-sections -mlongcalls -mtext-section-literals -mno-serialize-volatile   -o CMakeFiles/jerry-core.dir/__/jerry-all-in.c.obj   -c /home/wang/jerryscript/build/obj-esp8266/jerry-all-in.c
[... cut ...]
make[3]: Leaving directory '/home/wang/jerryscript/build/obj-esp8266'
CMakeFiles/Makefile2:152: recipe for target 'jerry-core/CMakeFiles/jerry-core.dir/rule' failed
make[2]: *** [jerry-core/CMakeFiles/jerry-core.dir/rule] Error 2
make[2]: Leaving directory '/home/wang/jerryscript/build/obj-esp8266'
Makefile:175: recipe for target 'jerry-core' failed
make[1]: *** [jerry-core] Error 2
make[1]: Leaving directory '/home/wang/jerryscript/build/obj-esp8266'
targets/esp8266/Makefile.esp8266:44: recipe for target 'jerry' failed
make: *** [jerry] Error 2

rerobika, 2017-11-22:
@wangnuannuan Are you completely sure that you followed the manual step by step? As I see the folder structure that you use differs from the one described in the manual. Maybe it could cause this error. Please try to follow the manual thoroughly and build the SDK again. Unfortunately I can't reproduce this error I'm only guessing what could be wrong.

zherczeg, 2018-01-29:
I think I close this issue. Please reopen if needed.

closed 2018-01-29 by zherczeg