jerryscript-project/jerryscript #2124: How call JavaScript function from C

zhaojuntao, 2017-11-28:
I want to call javascript function in *.js file from C   
example hello.js, containing func1, func2,  In C function, How do I call func1 in C files?

LaszloLango, 2017-11-28:
@zhaojuntao you can get your function objects with [`jerry_get_property`](http://jerryscript.net/api-reference/#jerry_get_property), then you can call them with [`jerry_call_function`](http://jerryscript.net/api-reference/#jerry_call_function)

zhaojuntao, 2017-11-28:
@LaszloLango 
That means I just need to read the script source , then  `jerry_parse` and  `jerry_run` ， finally i can use `jerry_get_property`  to get js file` function objects,  then call and execute it.

Is that right?
THANKS

LaszloLango, 2017-11-28:
@zhaojuntao yes

zhaojuntao, 2017-11-28:
@LaszloLango Thanks ！ I have got it, code is passed.

closed 2017-11-28 by zhaojuntao