jerryscript-project/jerryscript #2134: Parsing Problem for Regex containing {} brackets

billyauhk, 2017-12-04:
I was trying to put a code compiled with Emscripten and run them on both node.js and iot.js, however iot.js fails to parse a particular regular expression. After tracing the issue I think JerryScript is not-compliant with other ECMA Script implementations.

The line given by Emscripten is (matching a JavaScript function):
var sourceRegex = /^function\s*[a-zA-Z$_0-9]*\s*\(([^)]*)\)\s*{\s*([^*]*?)[\s;]*(?:return\s*(.*?)[;\s]*)?}$/;
The minified code which produces a parsing error in jerryscript is:
`var sourceRegex = /{}/;`

On other JavaScript engines (like v8, node.js and the script console on my Firefox), the above minified code passed the parsing, while in JerryScript (and thus iot.js), it results in a parsing error.

Section 15.10 of ECMA262 seems to allow the above regular expression as the "{" (or "}" character) can be interpreted as an Atom instead of a start of QuantifierPrefix:
https://www.ecma-international.org/ecma-262/5.1/#sec-15.10

galpeter, 2017-12-04:
Hi!
I've checked the standard and for me it seems that the Atom should not accept the "{" character (as the PatternCharacter can be a SourceCharacter but not any of `^ $ \ . * + ? ( ) [ ] { } |`) thus the engine is ECMA compliant. Still that is true that other engines allow this extra element.

zherczeg, 2017-12-05:
ES6 maybe?

billyauhk, 2017-12-05:
Thanks @galpeter for a deeper look into the standard. I now agree that the current implementation is ECMA compliant (and other engines implement this as sort of extension).

(Re @zherczeg ) Seems ES6 also excluded "{" as an Atom (Atom -> PatternCharacter -> SourceCharacter but not SyntaxCharacter), and SyntaxCharacter is simply the same list as in ES5.1.

The standard tests for ECMA262 also use the escaped versions:
https://github.com/tc39/test262/blob/master/harness/nativeFunctionMatcher.js

However, I would like to know if such an extension could be provided upon giving JerryScript a setting (either in build-time or as a runtime argument) that provides the compatibility with other engines? This may up to active developers as JerryScript should be kept small to fit in the embedded systems.

Maybe the developers could:
1) Close the issue; or
2) Re-label this as a feature request.

galpeter, 2018-01-22:
@billyauhk @szledan was kind enough to create a feature allow parsing `{` characters. Please see: #2169

zherczeg, 2018-01-24:
Patch landed, closing bug.

closed 2018-01-24 by zherczeg