jerryscript-project/jerryscript #2145: Parsing problem in object expression

niklauslee, 2017-12-09:
I've found a strange parsing problem in object expression.

```
jerry> { a: 1, b: "abc" }
Script Error: SyntaxError
```

The above expression results a parsing error. But, the below works well.

```
jerry> var x = { a: 1, b: "abc" };
undefined
```

One more strange thing is the following:

```
jerry> { a: 1 }
1
```

This should results "[object Object]", isn't it?

zherczeg, 2017-12-10:
Actually this is the correct behaviour.

By default '{' starts a code block, not an object. You need to enclose the object in brackets:

`{{ a: 1, b: "abc" }}`

`{ a: 1 }` In this case you created a code block with a single labelled statement. `{ 1:1 }` this would be a parse error, although a correct object definition.

niklauslee, 2017-12-11:
Thank you for your prompt answer :)

But, in Node.js REPL and Chrome Browser's console, I got the expected results like below:

```
$ node
> { a: 1 }
{ a: 1 }
> { a: 1, b: "abc" }
{ a: 1, b: 'abc' }
> 
```

I have some more questions about this:
1. Jerryscript has different syntax rules compared to V8? (Why code block has higher priority than object expession?)
2. This syntax rules are ECMA 5.1 standard compliant?

Thank you in advance.

zherczeg, 2017-12-11:
Lets start with the standard.

5.0:
https://www.ecma-international.org/ecma-262/5.1/#sec-12.4

Expression is something which does not start with `{` or `function`.

It even explains the reason:  An ExpressionStatement cannot start with an opening curly brace because that might make it ambiguous with a Block. Also, an ExpressionStatement cannot start with the function keyword because that might make it ambiguous with a FunctionDeclaration.

6.0:
https://www.ecma-international.org/ecma-262/6.0/#sec-expression-statement

Same, just `class` and `let` added as well, and the explanation is much longer.

Firefox also throws a syntax error.
`{ a: 1, b: "abc" }`
`SyntaxError: missing ; before statement`

Chrome probably has a bug, since `{ a:1 }` is ambiguous. Please report them this issue.

niklauslee, 2017-12-11:
OK. Now I understand it. Thank you!

closed 2017-12-11 by niklauslee