jerryscript-project/jerryscript #2148: CI fialed in Mbed/K64F build

jiangzidong, 2017-12-13:
the target `Mbed/K64F` failed in CI. seems the URL error or the version conflict

```
removed ‘/etc/apt/sources.list.d/basho_riak.list’
Executing: /tmp/tmp.qLWGriNsXE/gpg.1.sh --keyserver
hkp://keyserver.ubuntu.com:80
--recv
EA312927
gpg: requesting key EA312927 from hkp server keyserver.ubuntu.com
gpg: key EA312927: public key "MongoDB 3.2 Release Signing Key <packaging@mongodb.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1  (RSA: 1)
W: The repository 'http://ppa.launchpad.net/couchdb/stable/ubuntu trusty Release' does not have a Release file.
W: Failed to fetch https://packagecloud.io/computology/apt-backport/ubuntu/dists/trusty/InRelease  Failed to connect to packagecloud.io port 443: Connection timed out
W: Failed to fetch http://repo.mongodb.org/apt/ubuntu/dists/trusty/mongodb-org/3.4/InRelease  Could not connect to repo.mongodb.org:80 (52.222.215.211), connection timed out
W: Failed to fetch http://ppa.launchpad.net/openjdk-r/ppa/ubuntu/dists/trusty/InRelease  Could not connect to ppa.launchpad.net:80 (91.189.95.83), connection timed out
W: Failed to fetch http://ppa.launchpad.net/pollinate/ppa/ubuntu/dists/trusty/InRelease  Unable to connect to ppa.launchpad.net:http:
W: Failed to fetch http://ppa.launchpad.net/webupd8team/java/ubuntu/dists/trusty/InRelease  Unable to connect to ppa.launchpad.net:http:
W: Failed to fetch http://ppa.launchpad.net/chris-lea/redis-server/ubuntu/dists/trusty/main/binary-amd64/Packages  Unable to connect to ppa.launchpad.net:http:
W: Failed to fetch http://ppa.launchpad.net/chris-lea/redis-server/ubuntu/dists/trusty/main/binary-i386/Packages  Unable to connect to ppa.launchpad.net:http:
W: Failed to fetch http://ppa.launchpad.net/chris-lea/redis-server/ubuntu/dists/trusty/main/i18n/Translation-en  Unable to connect to ppa.launchpad.net:http:
W: Failed to fetch http://ppa.launchpad.net/git-core/ppa/ubuntu/dists/trusty/main/binary-amd64/Packages  Unable to connect to ppa.launchpad.net:http:
W: Failed to fetch http://ppa.launchpad.net/git-core/ppa/ubuntu/dists/trusty/main/binary-i386/Packages  Unable to connect to ppa.launchpad.net:http:
W: Failed to fetch http://ppa.launchpad.net/git-core/ppa/ubuntu/dists/trusty/main/i18n/Translation-en  Unable to connect to ppa.launchpad.net:http:
W: Failed to fetch http://ppa.launchpad.net/couchdb/stable/ubuntu/dists/trusty/main/binary-amd64/Packages  Unable to connect to ppa.launchpad.net:http:
W: Failed to fetch http://ppa.launchpad.net/couchdb/stable/ubuntu/dists/trusty/main/binary-i386/Packages  Unable to connect to ppa.launchpad.net:http:
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

```
$ make -f ./targets/mbed/Makefile.travis install
sudo apt-get install -q -y ninja-build libffi-dev libssl-dev
Reading package lists...
Building dependency tree...
Reading state information...
libffi-dev is already the newest version (3.1~rc1+r3.0.13-12ubuntu0.2).
libssl-dev is already the newest version (1.0.1f-1ubuntu2.23).
libssl-dev set to manually installed.
The following NEW packages will be installed:
  ninja-build
0 upgraded, 1 newly installed, 0 to remove and 15 not upgraded.
Need to get 79.6 kB of archives.
After this operation, 317 kB of additional disk space will be used.
Get:1 http://us-central1.gce.archive.ubuntu.com/ubuntu trusty-updates/main amd64 ninja-build amd64 1.3.4-1.1ubuntu0.14.04.1 [79.6 kB]
Fetched 79.6 kB in 0s (4,697 kB/s)
Selecting previously unselected package ninja-build.
(Reading database ... 101322 files and directories currently installed.)
Preparing to unpack .../ninja-build_1.3.4-1.1ubuntu0.14.04.1_amd64.deb ...
Unpacking ninja-build (1.3.4-1.1ubuntu0.14.04.1) ...
Processing triggers for man-db (2.6.7.1-1ubuntu1) ...
Setting up ninja-build (1.3.4-1.1ubuntu0.14.04.1) ...
pyenv global 2.7.13 # force the python version to a newer one
pyenv: version `2.7.13' not installed
make: *** [install-yotta] Error 1
The command "make -f ./targets/mbed/Makefile.travis install" failed and exited with 2 during .
Your build has been stopped.
```

galpeter, 2017-12-18:
Temporarily we'll allow the mbed failures ( #2151 ) working on a solution already.

akosthekiss, 2018-05-18:
The CI job has been fixed by #2333 but the whole target may soon be removed according to #2338 .

closed 2018-05-18 by akosthekiss