jerryscript-project/jerryscript #2149: Remove mbedos5 target

janjongboom, 2017-12-13:
After the recent conversation on DigitalIn and friends, I think we need to change how the Mbed OS and JerryScript integrate. Having part of the build toolchain in jerryscript core repository does not make sense, and things like generating `main.cpp` and the `Makefile` from here are also annoying for users, as they'd need to dive into the jerryscript repository to change these. Additionally, the Mbed.js build system is now different from the normal Mbed OS build system, which adds overhead.

So in the next few weeks I plan to open a PR to remove the mbedos5 target from JerryScript. 

Current development for the new system is at https://github.com/ARMmbed/mbed-js-v2 - this no longer requires mbedos5 target in JerryScript to be present.

zherczeg, 2018-01-29:
Any progress on this?

janjongboom, 2018-01-29:
@zherczeg We have an ongoing discussion internally about the new structure. So this is on hold for now.

akosthekiss, 2019-01-11:
Pinging this issue. It's been almost a year now since the last comment.

My two cents: I think that the original suggestion was to move sources from targets/mbedos5/jerryscript-mbed into another repo, which is OK with me. But it would be great if some wrapper / driver / stub (you name it) stayed here together with some makefile (Makefile and/or Makefile.travis) pulling / cloning that new repo and testing integration on JerryScript's side as well.

janjongboom, 2019-01-11:
@akosthekiss Yeah, we have diverged from this idea for now.

closed 2019-01-11 by janjongboom