jerryscript-project/jerryscript #2156: How to define getter/setter for array-like indexing?

niklauslee, 2017-12-26:
I want to define `Buffer` class like in Node.js. According to the Node.js docs, a buffer object can be used like as an array. For example, you can read a byte data at the first index by `buffer[0]`.

I'm trying to implement Buffer object in native C and I found `jerry_define_own_property`. But I don't know how to define getter for array-link indexing. Is it possible to define getter/setter for array-like indexing? If so, how to do it?

galpeter, 2018-01-22:
Currently JerryScript does not support user defined indexing. That would require something like ES6 Proxy objects (imho).

However for the Buffer object you could use the TypedArray parts from JerryScript which gives you the indexing capability for an underlying arraybuffer. You could create your own object and use the Uint8Array as a prototype for it to get the indexing operation.

niklauslee, 2018-01-23:
@galpeter Thank you for answering. I'm trying to use ArrayBuffer/TypedArray instead of implementing Buffer class.

akosthekiss, 2018-05-10:
This issue has been inactive for a while, probably because the question seems to have been answered. Closing.

(@niklauslee feel free to reopen the issue if the suggested approach did not work out at the end.)

closed 2018-05-10 by akosthekiss