jerryscript-project/jerryscript #2170: Keep resource name for function objects?

martijnthe, 2018-01-23:
Hey JrS people!

I'm trying to implement a native binding much alike Node's [`require.resolve()`](https://nodejs.org/api/modules.html#modules_require_resolve_request_options)
The tricky thing about the `require.resolve` function is that when using a relative path, i.e. `./thing.js` the look up happens relative to the source file in which the `require.resolve()` call was written.

I noticed that for the JerryScript debugger support, `jerry_parse_named_resource` was added, which seemed promising at a glance. Unfortunately, the "resource name" is merely passed off to the debugger client and not kept around.

Any suggestions on how to solve this?

I guess I want an API like `const jerry_char_t *jerry_get_current_resource_name(void);` that would return the currently executing resource name (or `NULL` if no named resource is executing / it's not available).

For reference, V8 has a [`GetScriptName()`](https://v8docs.nodesource.com/node-9.3/dd/d1e/classv8_1_1_stack_frame.html#a63d53e7637627027f0fe77e0e9dbeab7) method on the `StackFrame` class which is similar to what I'm looking for.

zherczeg, 2018-01-24:
Easy to keep them, but it consumes memory. The question is how can we do this effectively.

zherczeg, 2018-01-24:
Perhaps a new feature flag will be needed.

martijnthe, 2018-02-05:
I think to be able to implement Node.js' `__filename` you'd also need to have this API.

PR #3236 (2019-11-06): Introduce jerry_get_resource_name API function
This new API function adds possibility to query the resource name of the currently executed script (including modules) or a function object.
This patch closes #2170.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
commit 525c35f1 (2019-11-06): Introduce jerry_get_resource_name API function (#3236)

This new API function adds possibility to query the resource name of the currently executed script (including modules) or a function object.
This patch closes #2170.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
closed 2019-10-21 by rerobika