jerryscript-project/jerryscript #2173: Is there a jerry_string_to_char_buffer() that will fill a buffer, even if jerry string is too big?

mofosyne, 2018-01-24:
Specifically this check `(ecma_string_get_size (str_p) > buffer_size)` is exiting as expected if the jerry value is larger than the buffer.

However it would be nice if there was another function that would be more tolerant of too big jerry string, by simply dropping the extra characters.

I'm not too sure on the best option of dealing with this. Should this be a seperate function like `jerry_string_to_char_buffer_unsafe()`?

Note: If you are approaching this, then also check if ecma_string_copy_to_cesu8_buffer() actually checks buffer_size(), as memcpy uses the jerry string size rather than declared buffer size (or the smallest of either). 

https://github.com/jerryscript-project/jerryscript/blob/master/jerry-core/api/jerry.c#L1489

mofosyne, 2018-01-24:
The other alternative is a way to change the length of the jerry string temporarily (without copying the value), before placing it into jerry_string_to_char_buffer(). However I'm not aware of the best approach to do this.

mofosyne, 2018-01-24:
Hmmm... I solved my issue by using jerry_substring_to_char_buffer() using this:

```
  // Copy to buffer
  message_size = jerry_substring_to_char_buffer (message,                               /**< input string value */
                                                0,                                      /**< position of the first character */
                                                jerry_get_string_size(message),      /**< position of the last character */
                                                (unsigned char *)logger->temp_buffer,   /**< [out] output characters buffer */
                                                (sizeof(logger->temp_buffer) - 1)       /**< size of output buffer */
                                                );

```

We can close this ticket. But you may want to consider if a seperate function should be included to encourage usage if anyone needs a permissive version of jerry_string_to_char_buffer()

e.g.

```
jerry_size_t
jerry_string_to_char_buffer_permissive (const jerry_value_t value, /**< input string value */
                                jerry_char_t *buffer_p, /**< [out] output characters buffer */
                                jerry_size_t buffer_size) /**< size of output buffer */
{
    return jerry_substring_to_char_buffer (value, 0, jerry_get_string_size(message), buffer_p, buffer_size);
}
```

LaszloLango, 2018-01-24:
@mofosyne as you wrote `jerry_substring_to_char_buffer` should be used in your use case.

mofosyne, 2018-01-24:
Ah okay then, then maybe a documentation tip to consider jerry_substring_to_char_buffer() in those use case in jerry_string_to_char_buffer() and the uint8 case could be a good idea. I'll do a pull request of my documentation recommendation.

mofosyne, 2018-01-24:
https://github.com/jerryscript-project/jerryscript/pull/2174

closed 2018-02-08 by zherczeg