jerryscript-project/jerryscript #2175: Using C11 safe (_s) libc functions?

martijnthe, 2018-01-25:
How do you guys feel about adding the "safe" variants of libc functions, for example `memcpy_s`, `strcpy_s` etc? In our project, we're trying to eliminate use of the "unsafe" variants.

I would expect the code size to go up a little bit because of the additional argument(s).

So perhaps it should be a config `#define`?
OTOH, I think replacing the "unsafe" ones with "safe" ones would I think would avoid accidents and also avoid having `#define`s all over the place.

I imagine to add implementations to `jerry-libc`, I suspect some toolchains may not support these C11 additions yet. Having them in `jerry-libc` would be an easy compatibility path for those cases.

Thoughts? @akosthekiss @zherczeg

zherczeg, 2018-01-25:
In what sense they are safe?

martijnthe, 2018-01-25:
The `_s` functions take buffer sizes so that bounds checking can be done in those functions, instead of having to write logic to check the bounds before making a call to a non-`_s`/"unsafe" function.

For example:
```
size_t strnlen_s( const char *str, size_t strsz );
```

It's basically the same as `strlen`, except that it will stop at `str + strsz` to avoid running outside of the buffer.

See http://www.open-std.org/jtc1/sc22/wg14/www/docs/n1967.htm

zherczeg, 2018-01-25:
Hm, we only use strlen very few times (except tests), and i am not sure the buffer size is available there. Anyway I am not against it if it provides meaningful advantage.

martijnthe, 2018-01-25:
`strlen` was just an example. There are others like `memcpy_s`, `memcmp_s`, `memmove_s`, etc.

galpeter, 2018-02-01:
It would be good to use there safe functions. There are a few points to consider imho:
* Would this mean we would require C11 as our C standard (in case the jerry-libc is not used)?
* Also by adopting the safe variants would the jerry-libc contain the non-safe variants?
* What would be the code-size and performance impact? I know it is impossible to say at this stage, but maybe someone have any info/experience about this?

akosthekiss, 2018-05-18:
FYI: the removal of jerry-libc is under consideration in #2332, so adding safe function variants to jerry-libc might not be an option.

rerobika, 2019-10-22:
The idea looks promising and #3234 has landed thanks to this idea. We can wrap the safe/unsafe function call into macros but the only problem is that in the current code base we should pass extra parameters to several functions to check the size of the destination buffer. That is a bit costly also these arguments should be hidden with a macros since not all targets support the C11 standard. Furthermore jerry-libc has already been removed, so IMO we can close this issue for now, please reopen it anytime if needed.

closed 2019-10-22 by rerobika