jerryscript-project/jerryscript #2189: Question: Calling function from snapshots

slaff, 2018-02-01:
I have the latest JerryScript from the master branch. And tried to do something that worked some time ago. Namely - execute a snapshot and from the snapshot call a function. 
Here are the steps that I have used:

1. I have a `tests.js`  with the following content
```js
function sysloop() {
  // empty function... 
};

print("main js OK");
```
2. I have compiled the `jerry-snapshot` binary using the minimal profile. 
3. Then I created a snapshot using the command below:
```
./build/bin/jerry-snapshot generate -o tests.js.snap tests.js
```
Which produced a file with the following data in it:

```
00000000  4a 52 52 59 08 00 00 00  00 00 00 00 48 00 00 00  |JRRY........H...|
00000010  01 00 00 00 18 00 00 00  04 00 01 00 01 00 02 00  |................|
00000020  00 02 03 04 01 00 04 00  06 00 04 00 42 00 03 29  |............B..)|
00000030  01 02 bd 46 00 00 00 00  02 00 01 00 21 00 00 00  |...F........!...|
00000040  00 00 00 00 46 00 00 00  24 00 00 00 07 00 73 79  |....F...$.....sy|
00000050  73 6c 6f 6f 70 00 00 00  05 00 70 72 69 6e 74 00  |sloop.....print.|
00000060  0a 00 6d 61 69 6e 20 6a  73 20 4f 4b              |..main js OK|
0000006c
```

5. After that I tried to load the snapshot and run a function from it using the code below:
```C
	// check the arguments 1 - snapshotname, 2 - function to call
	if(argc != 3) {
		printf("Usage:\n\t%s <snapshot-file> <function-name>", argv[0]);
		return 1;
	}

	// load the snapshot
	const uint32_t *snapshot_p = read_file (fileName, &snapshotSize);

	if (snapshot_p == NULL)
	{
		printf("Snapshot file load error");
		return 2;
    }

	// Initialize the JS VM
	jerry_init(JERRY_INIT_EMPTY);

	// execute the snapshot
	ret_value = jerry_exec_snapshot (snapshot_p, snapshotSize, true);
	if (jerry_value_has_error_flag (ret_value))
	{
		jerry_release_value(ret_value);
		printf("Unable to execute the snapshot");
		return 3;
	}

	jerry_release_value(ret_value);

	// run the function
	jerry_value_t global_obj_val = jerry_get_global_object();
	jerry_value_t prop_name_val = jerry_create_string((const jerry_char_t *) functionName);
	jerry_value_t jsFunction = jerry_get_property(global_obj_val, prop_name_val);
	jerry_release_value(prop_name_val);

	if (jerry_value_has_error_flag(jsFunction)) {
		debugf("Error: '%s' not defined!", functionName);
		jerry_release_value(jsFunction);
		jerry_release_value(global_obj_val);
		return false;
	}

	if (!jerry_value_is_function(jsFunction)) {
		debugf("Error: '%s' is not a function!", functionName);
		jerry_release_value(jsFunction);
		jerry_release_value(global_obj_val);
		return 4;
	}

	int ret_code = 0;
	jerry_value_t res = jerry_call_function(jsFunction, global_obj_val, NULL, 0);
	if (jerry_value_has_error_flag(res)) {
		ret_code = 5;
	}
```

And the test C application above is compiled with the same flags as the minimal profile
```sh
gcc -std=c99 -Wall -Os -g -O2 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions \
       -DCONFIG_DISABLE_BUILTINS -DCONFIG_DISABLE_ES2015 -DCONFIG_DISABLE_UNICODE_CASE_CONVERSION -DCONFIG_DISABLE_DATE_BUILTIN -DCONFIG_DISABLE_STRING_BUILTIN \
       -DCONFIG_DISABLE_REGEXP_BUILTIN -DCONFIG_DISABLE_TYPEDARRAY_BUILTIN -DCONFIG_ECMA_NUMBER_TYPE=CONFIG_ECMA_NUMBER_FLOAT32 -DCONFIG_MEM_HEAP_AREA_SIZE=512 \
       -DCONFIG_ECMA_LCACHE_DISABLE -DJERRY_DISABLE_JS_PARSER -DJERRY_ENABLE_SNAPSHOT_EXEC -DJERRY_NDEBUG -D_BSD_SOURCE \
       -D__attr_always_inline___= -DJERRY_CONST_DATA="" -Ijerry-core/ -Ijerry-core/jmem -Ijerry-core/lit -Ijerry-core/ecma -Ijerry-core/ecma/base \
       -Ijerry-core/ecma/operations -Ijerry-core/ecma/builtin-objects -Ijerry-core/ecma/builtin-objects/typedarray -Ijerry-core/parser \
       -Ijerry-core/parser/regexp -Ijerry-core/parser/js -Ijerry-core/jrt -Ijerry-core/vm -Ijerry-core/api -Ijerry-core/include -Ijerry-core/profiles \
       -Ijerry-core/jcontext -Ijerry-core/debugger \
       -c js-snapshot.c
```

BUT the code runs until it reaches the following condition
```
if (!jerry_value_is_function(jsFunction)) {
		debugf("Error: '%s' is not a function!", functionName);
```

and reports that "sysloop" is not a function? 

Therefore I would be very thankful if someone can explain me what am I doing wrong? Again, this is a code that used to work with older versions (at the times when the snapshot save format was having v 6 (now it is v8) )

galpeter, 2018-02-01:
@slaff Hi, I've tried a few variations trying to reproduce your error but was unable to do so (tried multiple git commits also/full and minimal profile). I've created the exact same js snapshot binary and used your C code parts (extended with adding the print handler) however, it ran as expected (printed out the message).

What we could do is check what is the type of the `jsFunction` with the `jerry_value_is_...` calls. Additionally could you perhaps share the git hash and compiler version you were using?

galpeter, 2018-02-01:
Ok, so I've tried another think: recompiled the library without snapshot exec support and then I get the same error you are reporting.
You could check if the jerry lib you are using have snapshot exec support via the `jerry_is_feature_enabled (JERRY_FEATURE_SNAPSHOT_EXEC)` call.

slaff, 2018-02-01:
@zherczeg @galpeter Thank you guys for all the help and the efforts. I was using relatively old very of JerryScript where the "print" function was automatically registered. With the latest version this is no longer the case. I made a change and added, right after the initialization, also the registration for the "print" function and things started to work as expected.

Before I close this ticket I would like to ask one question: if one of the operations for the value objects fails is there a way to see a human readable explanation of why it failed? Something like: 
```C
jerry_error_explain(jsFunction);
```

zherczeg, 2018-02-02:
If error messages are on, the error has a description. It is in the `message` property. You can also convert the error to string.

slaff, 2018-02-02:
Thanks a lot for the awesome support!

closed 2018-02-02 by slaff

galpeter, 2018-02-02:
Currently there is no one-function way to get the error message. What is in progress now is a way to get the error type (type/reference/etc) (one of the open PRs currenlty).