jerryscript-project/jerryscript #2190: stack-overflow in re_match_regexp

renatahodovan, 2018-02-01:
###### Jerry version:

```
Checked revision: 918eb22a
Build command: ./tools/build.py --clean --debug --compile-flag=-fsanitize=address --compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer --compile-flag=-fno-common --compile-flag=-g --jerry-libc=off --static-link=off --strip=off --system-allocator=on --linker-flag=-fuse-ld=gold --error-messages=on --profile=es2015-subset
```
###### OS:

```
Ubuntu 17.10
```
###### Test case:

```javascript
/(?:(?=x)){1000}xyz/.exec('xyz'); 
```

Usually, I do not report stack-overflows since they are due to recursive js calls mostly.
However, in this case the failure disappears if I remove either the positive lookahed or the non-capturing brackets.
So I leave to the experts to decide whether this is a feature or a bug.

###### Backtrace:

```
ASAN:DEADLYSIGNAL
=================================================================
==24793==ERROR: AddressSanitizer: stack-overflow on address 0xff0acfe8 (pc 0x080cead4 bp 0xff0ad448 sp 0xff0acfec T0)
    #0 0x80cead3 in __interceptor_memcpy.part.43 (jerryscript/build/bin/jerry+0x80cead3)
    #1 0x834eda7 in re_get_value jerryscript/jerry-core/parser/regexp/re-bytecode.c:157:3
    #2 0x82fb7a5 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:947:19
    #3 0x82fd375 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1032:38
    #4 0x82f5e53 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:565:27
    #5 0x82f9fc1 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:840:38
    #6 0x82fc69e in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:987:38
    #7 0x82fd375 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1032:38
    #8 0x82f5e53 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:565:27
    #9 0x82f9fc1 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:840:38
    #10 0x82fc69e in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:987:38
    #11 0x82fd375 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1032:38
    #12 0x82f5e53 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:565:27
    #13 0x82f9fc1 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:840:38
    #14 0x82fc69e in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:987:38
    #15 0x82fd375 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1032:38
    #16 0x82f5e53 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:565:27
    #17 0x82f9fc1 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:840:38
    #18 0x82fc69e in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:987:38
    #19 0x82fd375 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1032:38
    #20 0x82f5e53 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:565:27
    #21 0x82f9fc1 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:840:38
    #22 0x82fc69e in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:987:38
    #23 0x82fd375 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1032:38
    #24 0x82f5e53 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:565:27
    #25 0x82f9fc1 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:840:38
    #26 0x82fc69e in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:987:38
    #27 0x82fd375 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1032:38
    #28 0x82f5e53 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:565:27
    #29 0x82f9fc1 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:840:38
    #30 0x82fc69e in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:987:38
    #31 0x82fd375 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1032:38
    #32 0x82f5e53 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:565:27
    #33 0x82f9fc1 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:840:38
    #34 0x82fc69e in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:987:38
    #35 0x82fd375 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1032:38
    #36 0x82f5e53 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:565:27
    #37 0x82f9fc1 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:840:38
    #38 0x82fc69e in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:987:38
    #39 0x82fd375 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1032:38
    #40 0x82f5e53 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:565:27
    #41 0x82f9fc1 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:840:38
    #42 0x82fc69e in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:987:38
[... 189 lines cut ...]
    #232 0x82f5e53 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:565:27
    #233 0x82f9fc1 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:840:38
    #234 0x82fc69e in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:987:38
    #235 0x82fd375 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1032:38
    #236 0x82f5e53 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:565:27
    #237 0x82f9fc1 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:840:38
    #238 0x82fc69e in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:987:38
    #239 0x82fd375 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1032:38
    #240 0x82f5e53 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:565:27
    #241 0x82f9fc1 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:840:38
    #242 0x82fc69e in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:987:38
    #243 0x82fd375 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1032:38
    #244 0x82f5e53 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:565:27
    #245 0x82f9fc1 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:840:38
    #246 0x82fc69e in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:987:38
    #247 0x82fd375 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:1032:38
    #248 0x82f5e53 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:565:27
    #249 0x82f9fc1 in re_match_regexp jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:840:38

SUMMARY: AddressSanitizer: stack-overflow (jerryscript/build/bin/jerry+0x80cead3) in __interceptor_memcpy.part.43
==24793==ABORTING

```

<sup>Found by [Fuzzinator](http://fuzzinator.readthedocs.io/) with [grammarinator](https://github.com/renatahodovan/grammarinator). </sup>

LaszloLango, 2018-02-16:
I cannot reproduce this issue.

renatahodovan, 2018-02-16:
It seems that it's only reproducible with a clang build:

CC=clang ./tools/build.py --clean --debug --compile-flag=-fsanitize=address --compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer --compile-flag=-fno-common --compile-flag=-g --jerry-libc=off --static-link=off --strip=off --system-allocator=on --linker-flag=-fuse-ld=gold --lto=off --profile=es2015-subset --error-messages=on

PR #2543 (2019-01-17): Add RegExp recursion depth limit
The regexp engine does not have any recursion depth check, thus it can cause problems with various regexps. Added a new build option `--regexp-recursion-limit N` whose
default value is 0, which is for unlimited recursion depth. Also added a build-option-test.

Fixes #2448
Fixes #2190

JerryScript-DCO-1.0-Signed-off-by: Istvan Miklos imiklos2@inf.u-szeged.hu
commit c23cf417 (2019-01-17): Add RegExp recursion depth limit (#2543)

The regexp engine does not have any recursion depth check, thus it can cause problems with various regexps. Added a new build option `--regexp-recursion-limit N` whose
default value is 0, which is for unlimited recursion depth. Also added a build-option-test.

Fixes #2448
Fixes #2190

JerryScript-DCO-1.0-Signed-off-by: Istvan Miklos imiklos2@inf.u-szeged.hu
closed 2019-01-17 by akosthekiss