jerryscript-project/jerryscript #2193: [ERROR]when runing example, undefined reference to `jerry_run_simple'

zhvala, 2018-02-01:
I tried to run example on ubuntu 14.04. I got a demo.c

```
#include <string.h>
#include "jerryscript.h"

int
main (void)
{
  const jerry_char_t script[] = "var str = 'Hello, World!';";
  size_t script_size = strlen ((const char *) script);

  bool ret_value = jerry_run_simple (script, script_size, JERRY_INIT_EMPTY);

  return (ret_value ? 0 : 1);
}
```
then run
```
$ gcc -I/usr/local/include -L/usr/local/lib -o demo demo.c
demo.c:(.text+0x68): undefined reference to `jerry_run_simple'
collect2: error: ld returned 1 exit status
```
I have got `.h` and `.a` files in my `/use/local/include` and `/usr/local/lib` path.

galpeter, 2018-02-01:
I think you'll need to link to the jerry library, which jerry archives do you have in the /use/local/lib path?
Probably you can use the `-ljerry-core -ljerry-libm -ljerry-port-default` options to resolve the link issue.

zhvala, 2018-02-01:
@galpeter 
```
$ root@34e1e81f6644:/usr/local/lib# ls
libjerry-core.a  libjerry-libm.a                  libjerry-port-default.a  python3.4
libjerry-ext.a   libjerry-port-default-minimal.a  python2.7
```
I add your options, not work unfortunately.

galpeter, 2018-02-01:
Could you share your full build command (with the `-l` options)?

 I've tried with the following one:
`gcc -I../installed/usr/local/include/ -L../installed/usr/local/lib/ -o demo demo.c -ljerry-core -ljerry-libm -ljerry-port-default` and the file was built without any problems.

(Note: the `../installed/` is my base dir for the jerry installation)

zhvala, 2018-02-01:
@galpeter
Thank you very much. It works.
It turn out I should add these options after my target not before it.
Thanks, again.

closed 2018-02-01 by LaszloLango

YaoHouyou, 2019-03-04:
I have the same problem on ubuntu 14.04，and tried  the above method，but failed.

The command is:
`gcc -I/usr/local/include/ -L/usr/local/lib/ -o demo demo.c -ljerry-core -ljerry-libm -ljerry-port-default`

I have got following .h and .a files in my `/use/local/include` and` /usr/local/lib` path.
`nisl@nisl-H81M-DS2:/usr/local/include$ls `
`jerryscript-compiler.h            jerryscript.h
jerryscript-core.h                jerryscript-port-default.h
jerryscript-debugger.h            jerryscript-port.h
jerryscript-debugger-transport.h  jerryscript-snapshot.h
jerryscript-ext                   math.h`

`nisl@nisl-H81M-DS2:/usr/local/lib$ ls`
`libjerry-core.a  libjerry-port-default.a          python3.4
libjerry-ext.a   libjerry-port-default-minimal.a
libjerry-libm.a  python2.7`

galpeter, 2019-03-04:
@YaoHouyou What is your compiler error? Based on your comment there is some kind of problem but without knowing the error it is hard to give assistance. Could you please share the compile log and the version of JerryScript you are using? In addition it would be good to have a separate issue report for it.

AbhinavMir, 2020-03-25:
Hi, what does "gcc -I../installed/usr/local/include/ -L../installed/usr/local/lib/" do?
New to GCC.

AbhinavMir, 2020-03-25:
Hi, what does "gcc -I../installed/usr/local/include/ -L../installed/usr/local/lib/" do?

galpeter, 2020-03-25:
Hi @AbhinavMir 

In short:
The `-I` option specifies the path where header/include files should be searched. E.g.: if `#include "file.h" is written in the C file, the compiler will also search for the "file.h" in the directories provided via the `-I` options.

The `-L` options is similar, just not for include files but for libraries during linking.

For more information on these please refer to the gcc/clang manuals.