jerryscript-project/jerryscript #2200: Heap buffer overflow while parsing malformed JSON hex escape sequence w/o closing quote

martijnthe, 2018-02-10:
Jerry version:

Checked revision: `e8608707b6d9486022b4d72b280303923945fdeb`
Build command: `./tools/build.py --clean --debug --compile-flag=-fsanitize=address --compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer --compile-flag=-fno-common --compile-flag=-g --jerry-libc=off --static-link=off --strip=off --system-allocator=on --error-messages=on --profile=es2015-subset`

OS:

macOS 10.12.6 (16G1212)

Test case:

```
JSON.parse('"\\ubad');
```

Backtrace:

```
==8770==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x0240155e at pc 0x001850dc bp 0xbffb4418 sp 0xbffb4414
READ of size 1 at 0x0240155e thread T0
    #0 0x1850db in lit_read_code_unit_from_hex lit-char-helpers.c:443
    #1 0x1e5412 in ecma_builtin_json_parse_string ecma-builtin-json.c:195
    #2 0x1e3821 in ecma_builtin_json_parse_next_token ecma-builtin-json.c:450
    #3 0x1e1fcb in ecma_builtin_json_parse_value ecma-builtin-json.c:565
    #4 0x10cdce in ecma_builtin_json_parse ecma-builtin-json.c:837
    #5 0x10c5cb in ecma_builtin_json_dispatch_routine ecma-builtin-json.inc.h:26
    #6 0x13d396 in ecma_builtin_dispatch_routine ecma-builtins.inc.h:135
    #7 0x13caff in ecma_builtin_dispatch_call ecma-builtins.c:844
    #8 0x7e3af in ecma_op_function_call ecma-function-object.c:342
    #9 0x22181f in opfunc_call vm.c:425
    #10 0x1dbd77 in vm_execute vm.c:2871
    #11 0x1572c4 in vm_run vm.c:2951
    #12 0x7e8df in ecma_op_function_call ecma-function-object.c:405
    #13 0x22181f in opfunc_call vm.c:425
    #14 0x1dbd77 in vm_execute vm.c:2871
    #15 0x1572c4 in vm_run vm.c:2951
    #16 0x55f82 in vm_run_global vm.c:232
    #17 0x53dc1 in jerry_run jerry.c:562
    #18 0x4d58f in main main-unix.c:611
    #19 0xa153c394 in start (libdyld.dylib:i386+0x5394)

0x0240155e is located 0 bytes to the right of 14-byte region [0x02401550,0x0240155e)
allocated by thread T0 here:
    #0 0x64752c in wrap_malloc (libclang_rt.asan_osx_dynamic.dylib:i386+0x5452c)
    #1 0x1fb416 in jmem_heap_alloc_block_internal jmem-heap.c:324
    #2 0x18450a in jmem_heap_gc_and_alloc_block jmem-heap.c:360
    #3 0x5e80e in ecma_new_ecma_string_from_utf8 jmem-heap.c:406
    #4 0xd3605 in ecma_find_or_create_literal_string ecma-literal-storage.c:73
    #5 0x202118 in parser_compute_indicies js-parser.c:201
    #6 0x1c2ae3 in parser_post_processing js-parser.c:1424
    #7 0x1c99ce in parser_parse_source js-parser.c:2229
    #8 0x540d8 in parser_parse_script js-parser.c:2726
    #9 0x5373d in jerry_parse jerry.c:388
    #10 0x55575 in jerry_parse_named_resource jerry.c:446
    #11 0x4d4a7 in main main-unix.c:602
    #12 0xa153c394 in start (libdyld.dylib:i386+0x5394)

SUMMARY: AddressSanitizer: heap-buffer-overflow lit-char-helpers.c:443 in lit_read_code_unit_from_hex
Shadow bytes around the buggy address:
  0x20480250: fa fa 00 04 fa fa 00 03 fa fa 00 03 fa fa 00 00
  0x20480260: fa fa 00 02 fa fa 00 02 fa fa 00 02 fa fa 00 00
  0x20480270: fa fa 00 06 fa fa 00 fa fa fa fd fd fa fa 00 05
  0x20480280: fa fa 00 00 fa fa 00 fa fa fa 00 fa fa fa fd fd
  0x20480290: fa fa 00 00 fa fa 00 06 fa fa fd fa fa fa 00 06
=>0x204802a0: fa fa fd fd fa fa 00 00 fa fa 00[06]fa fa 00 03
  0x204802b0: fa fa 00 03 fa fa 00 00 fa fa 00 01 fa fa 00 00
  0x204802c0: fa fa 00 02 fa fa 00 00 fa fa 00 05 fa fa fd fa
  0x204802d0: fa fa fd fd fa fa fd fa fa fa fd fd fa fa fd fd
  0x204802e0: fa fa fd fd fa fa fd fd fa fa fd fa fa fa fd fd
  0x204802f0: fa fa fd fd fa fa fd fa fa fa fd fd fa fa fd fd
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==8770==ABORTING
Abort trap: 6
```

martijnthe, 2018-02-10:
I'll post a fix shortly.

renatahodovan, 2018-02-10:
@martijnthe nice catch. I wish I found it :)

martijnthe, 2018-02-10:
@renatahodovan TBH I didn't find it myself either, clang's static analysis gave it away :D

closed 2018-02-16 by LaszloLango