jerryscript-project/jerryscript #2211: Windows port

PKarics, 2018-02-17:
Hello JerryScript,

I have a little question. :)

I'm comparing embeddable javascript engines. I need a js engine which is available on linux, macos, android, ios and windows.

I cannot see windows support now. Is it in progress or is it planned?

zherczeg, 2018-02-18:
JerryScript is a C application. You can compile it on any environment with a C compiler.

If you ask for windows build files, I am not aware that we have plans for it. Anyway it is very easy to do it: just add all .c and .h files from jerry-core subdirectory to your build. All-in-one build also works, so you can just create a `.c` file which includes all other `.c` files and add that single file to your build system. If you want to configure the engine, I recommend to check jerry-core/profiles/.

galpeter, 2018-02-22:
I managed to build Jerry on Windows with VS2015. There are a few things which needs a bit of fix for Windows and currently I've just thrown together "working solution". However it is also possible that using VS2017 and Clang (or Clang/C2) even less modification is required. Still I'll upload my modifications into my fork with a bit of description on how to build it.

PKarics, 2018-02-22:
Great. I am waiting for the modifications. 

I tried to compile the project with VS2017 15.5.6 and I stuck, because cannot solve linker issues about ecma_builtin_object_prototype_dispatch_construct  ecma_builtin_object_prototype_dispatch_call  functions. If I disable builtins then 6 linker errores come if not then 34.

I forked jerryscript on the weekend and pushed every modification to VS2017-15-5-6-compile branch. 
https://github.com/PKarics/jerryscript/tree/VS2017-15-5-6-compile

galpeter, 2018-02-22:
@PKarics You can overcome that link issue if you enable optimization (/O1) and disable runtime checks. At least that is also what I had to do.

akosthekiss, 2018-04-17:
@galpeter @PKarics This issue got closed but then the discussion continued. Has any conclusion or solution been reached? Should this issue be reopened? (Especially as #2285 also brought up the question of Windows support.)

lygstate, 2018-04-17:
I was trying building with vs 2017 with CMake but failed.

galpeter, 2018-04-18:
Sorry, not all fixes for windows are in the master yet. I'll get around to upload everything in this/next week.

lygstate, 2018-04-18:
@galpeter Oh, thanks a lot, anyway, compiling with CMake + clang should works:)

akosthekiss, 2018-04-26:
Reopening as discussion seems to continue here. I'll close the duplicate instead.

lygstate, 2018-05-02:
@PKarics No pull request? I wanna create one based on your changes.

lygstate, 2018-05-02:
@PKarics I've already created and also make some update:)

closed 2018-05-18 by yichoi

BluVueMatthew, 2019-03-25:
Question, is the windows support implemented? The Getting Started page says only Ubuntu is officially supported. There seems to be an awful hurry to close this item. However, an easier to setup implementation for Windows developers would go a long way towards adoption of JerryScript by those in the Microcontroller community who want to develop using JavaScript. Have you considered NPM or another package manager to provide the required packages? I ask as I spent the better part of the weekend working with Johnny Five only to realize their implementation requires most microcontrollers to be tethered to a pc via usb in order to operate, which seems to eliminate any of the use cases i have from being viable. I'd prefer not to spend a ton of time installing a bunch of packages just to get started or having to create a virtual machine to run Ubuntu.

LaszloLango, 2019-03-25:
@BluVueMatthew yes, JerryScript should work fine on Windows. It looks like we forgot to mention it in the related documentation.

galpeter, 2019-03-25:
@BluVueMatthew This PR was to track the JerryScript usage on Windows. Based on your comment you want to directly program a microcontroller, that is compile JerryScript on Windows an flash it on a target device. Is my assumption correct? If so, the cross compilation on Windows to a target device is not tested/supported yet. In addition, please note that this project is a JavaScript engine (just like v8/duktape) not a full framework like Johnny Five (node.js, etc..).

That being said, what would be your use case? Could you share a bit info about it?

matthewjlawr, 2019-04-09:
Thanks for following up! Sorry for delay in response, I'm experimenting with custom IOT devices, mostly arduino-compatible, such as node-mcu, esp-01, and various generic arduino boards(nano/mega/uno). Yes, your assumption is correct, I would like to write code in JavaScript and have that compiled into something the microcontroller understands, and flashed to it. The issue I ran into with Johnny 5 is that it did not support running directly on any of my boards, and required my boards be tethered via usb to a pc, which just does not make sense for and IOT ecosystem. I seem to have more ideas that time to complete them. I've modified some old roombas to run via mqtt commands over wifi with esp-01's and plan to "roll my own" web api on AWS as a learning exercise to manage them remotely. Another project I've gotten partially working is a wifi doorbell that monitors the voltage of my doorbell circuit, as I can't have a device mounted visibly on my apartment door. I've got 5-10 other projects, and it would be much easier to use something familiar like JavaScript.

xcjiang, 2022-01-26:
Is it possible to build jerryscript on cygwin?

ossy-szeged, 2022-01-27:
> Is it possible to build jerryscript on cygwin?

I agree with @LaszloLango, it might work, but we don't support it officially, we don't test it at all. I suggest Visual Studio + cmake in windows cmd: tools\build.py works fine and tested regularly.

PR #2306 (2018-05-18): Add Windows support
Fixes #2211