jerryscript-project/jerryscript #2230: heap-buffer-overflow in lit_read_code_unit_from_hex

zhunki, 2018-03-01:
samiliar bug with #2140

((new RegExp("[\\u0")).exec("u"));

=================================================================
==9567==ERROR: AddressSanitizer: heap-buffer-overflow on address 0xf580075c at pc 0x080f9f5a bp 0xffdf2a48 sp 0xffdf2a38
READ of size 1 at 0xf580075c thread T0
    #0 0x80f9f59 in lit_read_code_unit_from_hex /home/b/jerryscript/jerry-core/lit/lit-char-helpers.c:443
    #1 0x811df42 in re_parse_char_class /home/b/jerryscript/jerry-core/parser/regexp/re-parser.c:431
    #2 0x811a7b5 in re_parse_alternative /home/b/jerryscript/jerry-core/parser/regexp/re-compiler.c:399
    #3 0x811b5f7 in re_compile_bytecode /home/b/jerryscript/jerry-core/parser/regexp/re-compiler.c:564
    #4 0x80e2057 in ecma_op_create_regexp_object /home/b/jerryscript/jerry-core/ecma/operations/ecma-regexp-object.c:292
    #5 0x80bf83b in ecma_builtin_regexp_dispatch_construct /home/b/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-regexp.c:136
    #6 0x80a0e9d in ecma_builtin_dispatch_construct /home/b/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:154
    #7 0x81358b3 in opfunc_construct /home/b/jerryscript/jerry-core/vm/vm.c:475
    #8 0x81358b3 in vm_execute /home/b/jerryscript/jerry-core/vm/vm.c:2849
    #9 0x8137a0d in vm_run /home/b/jerryscript/jerry-core/vm/vm.c:2924
    #10 0x8137a0d in vm_run_global /home/b/jerryscript/jerry-core/vm/vm.c:224
    #11 0x8057509 in jerry_run /home/b/jerryscript/jerry-core/api/jerry.c:562
    #12 0x804c176 in main /home/b/jerryscript/jerry-main/main-unix.c:611
    #13 0xf7005636 in __libc_start_main (/lib/i386-linux-gnu/libc.so.6+0x18636)
    #14 0x804f8e9  (/home/b/jerryscript/build/bin/jerry+0x804f8e9)

0xf580075c is located 0 bytes to the right of 12-byte region [0xf5800750,0xf580075c)
allocated by thread T0 here:
    #0 0xf7239dee in malloc (/usr/lib32/libasan.so.2+0x96dee)
    #1 0x804f544 in jmem_heap_alloc_block_internal /home/b/jerryscript/jerry-core/jmem/jmem-heap.c:324
    #2 0x804f544 in jmem_heap_gc_and_alloc_block /home/b/jerryscript/jerry-core/jmem/jmem-heap.c:360
    #3 0x804f544 in jmem_heap_alloc_block /home/b/jerryscript/jerry-core/jmem/jmem-heap.c:406
    #4 0x80650aa in ecma_new_ecma_string_from_utf8 /home/b/jerryscript/jerry-core/ecma/base/ecma-helpers-string.c:232
    #5 0x807d593 in ecma_find_or_create_literal_string /home/b/jerryscript/jerry-core/ecma/base/ecma-literal-storage.c:73
    #6 0x810487a in parser_compute_indicies /home/b/jerryscript/jerry-core/parser/js/js-parser.c:201
    #7 0x810487a in parser_post_processing /home/b/jerryscript/jerry-core/parser/js/js-parser.c:1441
    #8 0x8110f8a in parser_parse_source /home/b/jerryscript/jerry-core/parser/js/js-parser.c:2267
    #9 0x8112c78 in parser_parse_script /home/b/jerryscript/jerry-core/parser/js/js-parser.c:2764
    #10 0x8056f72 in jerry_parse /home/b/jerryscript/jerry-core/api/jerry.c:388
    #11 0x8056f72 in jerry_parse_named_resource /home/b/jerryscript/jerry-core/api/jerry.c:446
    #12 0x804c140 in main /home/b/jerryscript/jerry-main/main-unix.c:602
    #13 0xf7005636 in __libc_start_main (/lib/i386-linux-gnu/libc.so.6+0x18636)

SUMMARY: AddressSanitizer: heap-buffer-overflow /home/b/jerryscript/jerry-core/lit/lit-char-helpers.c:443 lit_read_code_unit_from_hex
Shadow bytes around the buggy address:
  0x3eb00090: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x3eb000a0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x3eb000b0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x3eb000c0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x3eb000d0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
=>0x3eb000e0: fa fa 00 01 fa fa 00 00 fa fa 00[04]fa fa fd fa
  0x3eb000f0: fa fa 00 05 fa fa 00 02 fa fa 00 06 fa fa 00 00
  0x3eb00100: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x3eb00110: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x3eb00120: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x3eb00130: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Heap right redzone:      fb
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack partial redzone:   f4
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
==9567==ABORTING

renatahodovan, 2018-03-12:
I cannot reproduce this on the latest master (685af74a).

zhunki, 2018-03-15:
I can still reproduce on the latest master, maybe we use different build command? mine is 

python ./tools/build.py --clean --debug --compile-flag=-fsanitize=address --compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer --compile-flag=-fno-common --jerry-libc=off --static-link=off --lto=off --error-message=on --system-allocator=on

akosthekiss, 2018-04-27:
I cannot reproduce the issue either. @zhunki does it show up for you still? Below our my logs for current master for two different Ubuntu versions (with two different gccs):

```
$ cat /etc/issue
Ubuntu 14.04.5 LTS \n \l

$ python ./tools/build.py --clean --debug --compile-flag=-fsanitize=address --compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer --compile-flag=-fno-common --jerry-libc=off --static-link=off --lto=off --error-message=on --system-allocator=on
-- The C compiler identification is GNU 4.9.4
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- CMAKE_BUILD_TYPE          Debug
-- CMAKE_SYSTEM_NAME         Linux
-- CMAKE_SYSTEM_PROCESSOR    x86_64
-- ENABLE_ALL_IN_ONE         OFF
-- ENABLE_LTO                OFF
-- ENABLE_STATIC_LINK        OFF
-- ENABLE_STRIP              ON
-- JERRY_CMDLINE             ON
-- JERRY_CMDLINE_TEST        OFF
-- JERRY_CMDLINE_SNAPSHOT    OFF
-- JERRY_PORT_DEFAULT        ON (FORCED BY CMDLINE OR TESTS)
-- JERRY_EXT                 ON (FORCED BY CMDLINE OR TESTS)
-- JERRY_LIBC                OFF
-- JERRY_LIBM                ON
-- UNITTESTS                 OFF
-- DOCTESTS                  OFF
-- FEATURE_INIT_FINI         OFF
-- FEATURE_CPOINTER_32_BIT     ON (FORCED BY SYSTEM ALLOCATOR)
-- FEATURE_DEBUGGER            OFF
-- FEATURE_ERROR_MESSAGES      ON
-- FEATURE_EXTERNAL_CONTEXT    OFF
-- FEATURE_JS_PARSER           ON
-- FEATURE_LINE_INFO           OFF
-- FEATURE_MEM_STATS           OFF
-- FEATURE_MEM_STRESS_TEST     OFF
-- FEATURE_PARSER_DUMP         OFF
-- FEATURE_PROFILE             es5.1
-- FEATURE_REGEXP_STRICT_MODE  OFF
-- FEATURE_REGEXP_DUMP         OFF
-- FEATURE_SNAPSHOT_EXEC       OFF
-- FEATURE_SNAPSHOT_SAVE       OFF
-- FEATURE_SYSTEM_ALLOCATOR    ON
-- FEATURE_VALGRIND            OFF
-- FEATURE_VALGRIND_FREYA      OFF
-- FEATURE_VM_EXEC_STOP        OFF
-- MEM_HEAP_SIZE_KB            512
-- ENABLE_LINK_MAP           OFF
==============================
Build succeeded!
==============================
$ build/bin/jerry --version
Version: 1.0 (d672d1e)
$ cat issue-2230.js 
((new RegExp("[\u0")).exec("u"));
$ build/bin/jerry issue-2230.js 
((new RegExp("[\u0")).exec("u"));
~~~~~~~~~~~~~~~^
Script Error: SyntaxError: Invalid escape sequence. [line: 1, column: 16]
```

```
$ cat /etc/issue
Ubuntu 18.04 LTS \n \l

$ python ./tools/build.py --clean --debug --compile-flag=-fsanitize=address --compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer --compile-flag=-fno-common --jerry-libc=off --static-link=off --lto=off --error-message=on --system-allocator=on
-- The C compiler identification is GNU 7.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- CMAKE_BUILD_TYPE          Debug
-- CMAKE_SYSTEM_NAME         Linux
-- CMAKE_SYSTEM_PROCESSOR    x86_64
-- ENABLE_ALL_IN_ONE         OFF
-- ENABLE_LTO                OFF
-- ENABLE_STATIC_LINK        OFF
-- ENABLE_STRIP              ON
-- JERRY_CMDLINE             ON
-- JERRY_CMDLINE_TEST        OFF
-- JERRY_CMDLINE_SNAPSHOT    OFF
-- JERRY_PORT_DEFAULT        ON (FORCED BY CMDLINE OR TESTS)
-- JERRY_EXT                 ON (FORCED BY CMDLINE OR TESTS)
-- JERRY_LIBC                OFF
-- JERRY_LIBM                ON
-- UNITTESTS                 OFF
-- DOCTESTS                  OFF
-- FEATURE_INIT_FINI         OFF
-- FEATURE_CPOINTER_32_BIT     ON (FORCED BY SYSTEM ALLOCATOR)
-- FEATURE_DEBUGGER            OFF
-- FEATURE_ERROR_MESSAGES      ON
-- FEATURE_EXTERNAL_CONTEXT    OFF
-- FEATURE_JS_PARSER           ON
-- FEATURE_LINE_INFO           OFF
-- FEATURE_MEM_STATS           OFF
-- FEATURE_MEM_STRESS_TEST     OFF
-- FEATURE_PARSER_DUMP         OFF
-- FEATURE_PROFILE             es5.1
-- FEATURE_REGEXP_STRICT_MODE  OFF
-- FEATURE_REGEXP_DUMP         OFF
-- FEATURE_SNAPSHOT_EXEC       OFF
-- FEATURE_SNAPSHOT_SAVE       OFF
-- FEATURE_SYSTEM_ALLOCATOR    ON
-- FEATURE_VALGRIND            OFF
-- FEATURE_VALGRIND_FREYA      OFF
-- FEATURE_VM_EXEC_STOP        OFF
-- MEM_HEAP_SIZE_KB            512
-- ENABLE_LINK_MAP           OFF
==============================
Build succeeded!
==============================
$ build/bin/jerry --version
Version: 1.0 (d672d1e7)
$ cat issue-2230.js
((new RegExp("[\u0")).exec("u"));
$ build/bin/jerry issue-2230.js
((new RegExp("[\u0")).exec("u"));
~~~~~~~~~~~~~~~^
Script Error: SyntaxError: Invalid escape sequence. [line: 1, column: 16]
```

renatahodovan, 2018-05-19:
At the end, I've managed to reproduce the issue. It seems that the original test cases missed a backslash. The proper test case is

```js
((new RegExp("[\\u0")).exec("u"));
```

zhunki, 2018-05-25:
this is CVE-2018-11419.

akosthekiss, 2018-05-25:
Don't close until fixed

PR #2352 (2018-05-28): Fix heap buffer overflow in re_parse_char_class
This patch fixes #2230 and fixes #2237.
Test cases are added for both issues and also adds new cases which caused the same error.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
commit e6664f63 (2018-05-28): Fix heap buffer overflow in re_parse_char_class (#2352)

This patch fixes #2230 and fixes #2237.
Test cases are added for both issues and also adds new cases which caused the same error.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
closed 2018-05-28 by LaszloLango