jerryscript-project/jerryscript #2238: stack-overflow(recursion)

zhunki, 2018-03-11:
fn_expr = function NaN(a){
  'use strict';
   NaN(0);
}
fn_expr(1);

==1318==ERROR: AddressSanitizer: stack-overflow on address 0xff769fec (pc 0xf7ab9c45 bp 0xf7b0f000 sp 0xff769fe0 T0)
    #0 0xf7ab9c44  (/usr/lib32/libasan.so.2+0xb0c44)
    #1 0xf7ab9757  (/usr/lib32/libasan.so.2+0xb0757)
    #2 0xf7a246d2  (/usr/lib32/libasan.so.2+0x1b6d2)
    #3 0xf7a9fdbb in malloc (/usr/lib32/libasan.so.2+0x96dbb)
    #4 0x804f544 in jmem_heap_alloc_block_internal /home/b/jerryscript/jerry-core/jmem/jmem-heap.c:324
    #5 0x804f544 in jmem_heap_gc_and_alloc_block /home/b/jerryscript/jerry-core/jmem/jmem-heap.c:360
    #6 0x804f544 in jmem_heap_alloc_block /home/b/jerryscript/jerry-core/jmem/jmem-heap.c:406
    #7 0x808d5fc in ecma_create_object /home/b/jerryscript/jerry-core/ecma/base/ecma-helpers.c:99
    #8 0x80d3d01 in ecma_op_create_function_object /home/b/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:127
    #9 0x8135055 in vm_construct_literal_object /home/b/jerryscript/jerry-core/vm/vm.c:313
    #10 0x8135055 in vm_init_loop /home/b/jerryscript/jerry-core/vm/vm.c:637
    #11 0x8135055 in vm_execute /home/b/jerryscript/jerry-core/vm/vm.c:2831
    #12 0x81388ef in vm_run /home/b/jerryscript/jerry-core/vm/vm.c:2924
    #13 0x80d4a53 in ecma_op_function_call /home/b/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:405
    #14 0x8136396 in opfunc_call /home/b/jerryscript/jerry-core/vm/vm.c:417
    #15 0x8136396 in vm_execute /home/b/jerryscript/jerry-core/vm/vm.c:2844
    #16 0x81388ef in vm_run /home/b/jerryscript/jerry-core/vm/vm.c:2924
    #17 0x80d4a53 in ecma_op_function_call /home/b/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:405
    #18 0x8136396 in opfunc_call /home/b/jerryscript/jerry-core/vm/vm.c:417
    #19 0x8136396 in vm_execute /home/b/jerryscript/jerry-core/vm/vm.c:2844
    #20 0x81388ef in vm_run /home/b/jerryscript/jerry-core/vm/vm.c:2924
    #21 0x80d4a53 in ecma_op_function_call /home/b/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:405
    #22 0x8136396 in opfunc_call /home/b/jerryscript/jerry-core/vm/vm.c:417
    #23 0x8136396 in vm_execute /home/b/jerryscript/jerry-core/vm/vm.c:2844
    #24 0x81388ef in vm_run /home/b/jerryscript/jerry-core/vm/vm.c:2924
    #25 0x80d4a53 in ecma_op_function_call /home/b/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:405
    #26 0x8136396 in opfunc_call /home/b/jerryscript/jerry-core/vm/vm.c:417
    #27 0x8136396 in vm_execute /home/b/jerryscript/jerry-core/vm/vm.c:2844
    #28 0x81388ef in vm_run /home/b/jerryscript/jerry-core/vm/vm.c:2924
    #29 0x80d4a53 in ecma_op_function_call /home/b/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:405
    #30 0x8136396 in opfunc_call /home/b/jerryscript/jerry-core/vm/vm.c:417
    #31 0x8136396 in vm_execute /home/b/jerryscript/jerry-core/vm/vm.c:2844
    #32 0x81388ef in vm_run /home/b/jerryscript/jerry-core/vm/vm.c:2924
........

zherczeg, 2018-03-11:
Sounds like an expected behaviour. You can use compiler flags to check stack. ARM even reserves a register for it if you use it.

akosthekiss, 2018-04-27:
This issue report has its merits. I've created some variants of the reported test case and it seems that the behaviour of jerry varies depending on at least two factors: whether the name of the infinitely recursive function is a builtin and whether the body of the function uses strict semantics. Both seems to be necessary to trigger a stack overflow. If the function name is "generic" or its body is non-strict, the engine errors with `ERR_REF_COUNT_LIMIT`. It is also a fatal error, but the engine itself catches and reports it. I'm not sure that it is OK to have jerry behave so differently in the case of infinite recursion just because some seemingly independent factors change.

The test case variants I created and the results:
```
$ cat issue-2238-nan-strict.js 
f = function NaN() {
    'use strict';
    NaN();
}
f();
$ build/bin/jerry issue-2238-nan-strict.js 
ASAN:DEADLYSIGNAL
=================================================================
==5101==ERROR: AddressSanitizer: stack-overflow on address 0xff3d4ff0 (pc 0xf7ab1ea8 bp 0xff3d5418 sp 0xff3d4fd0 T0)
SUMMARY: AddressSanitizer: stack-overflow (/usr/lib32/libasan.so.4+0xe5ea7) in malloc
==5101==ABORTING

$ cat issue-2238-nan.js 
f = function NaN() {
    NaN();
}
f();
$ build/bin/jerry issue-2238-nan.js 
Error: ERR_REF_COUNT_LIMIT

$ cat issue-2238-g-strict.js 
f = function g() {
    'use strict';
    g();
}
f();
$ build/bin/jerry issue-2238-g-strict.js
Error: ERR_REF_COUNT_LIMIT

$ cat issue-2238-g.js 
f = function g() {
    g();
}
f();
$ build/bin/jerry issue-2238-g.js
Error: ERR_REF_COUNT_LIMIT

$ cat issue-2238-object-strict.js 
f = function Object() {
    'use strict';
    Object();
}
f();
$ build/bin/jerry issue-2238-object-strict.js 
ASAN:DEADLYSIGNAL
=================================================================
==5112==ERROR: AddressSanitizer: stack-overflow on address 0xff106ff8 (pc 0xf7961c55 bp 0xf7ab2000 sp 0xff106ff0 T0)
SUMMARY: AddressSanitizer: stack-overflow (/usr/lib32/libasan.so.4+0x1fc54) 
==5112==ABORTING

$ cat issue-2238-object.js 
f = function Object() {
    Object();
}
f();
$ build/bin/jerry issue-2238-object.js 
Error: ERR_REF_COUNT_LIMIT
```

Test environment:
```
$ cat /etc/issue
Ubuntu 18.04 LTS \n \l
$ python ./tools/build.py --clean --debug --compile-flag=-fsanitize=address --compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer --compile-flag=-fno-common --jerry-libc=off --static-link=off --lto=off --error-message=on --system-allocator=on
$ build/bin/jerry --version
Version: 1.0 (d672d1e7)
```

rerobika, 2019-10-16:
Let's finalize this issue.

#2969 has landed and now we are able to track stack-overflow internally.