jerryscript-project/jerryscript #2251: float vs double

DJLinux, 2018-03-24:
(Sorry about my bad english)
Hello I build JerryScript with CodeBlocks for Windows and Linux and Raspberry PI 32/ 64-bit 
With support for double "-d" and float "-f" so the resulting libs are:
JerryScript-32-f.dll, JerryScript-32-d.dll, JerryScript-64-f.dll, JerryScript-64-d.dll
libJerryScript-32-f.so, libJerryScript-32-d.so, libJerryScript-64-f.so, libJerryScript-64-d.so

Of course for the FLOAR32 build I disabled the JS date functionality and it works all so far.

But I wonder me that jerry_create_value and jerry_get_value are hard coded for "double" ?
I mean if I build JerryScript  for my retro computer I have only float my retro compiler 
don't have the data type double  at all.

So why does JerryScript  use double if 
(CONFIG_ECMA_NUMBER_TYPE == CONFIG_ECMA_NUMBER_FLOAT32)
are defined ?

Thank you DJ

zherczeg, 2018-03-25:
The public API cannot rely on compiler defines. In most environments, when you include a library header, you don't provide the list of compiler defines.

The solution I can suggest is redefine double with float in your computer environment. If you don't have the type, it should not cause any problems.

DJLinux, 2018-03-30:
#if (CONFIG_ECMA_NUMBER_TYPE == CONFIG_ECMA_NUMBER_FLOAT64)

x = TO_ECMA_NUMBER_T (fabs (x));
x = TO_ECMA_NUMBER_T (acos (x));
x = TO_ECMA_NUMBER_T (asin (x));
x = TO_ECMA_NUMBER_T (atan (x));
x = TO_ECMA_NUMBER_T (ceil (x));
x = TO_ECMA_NUMBER_T (cos (x));
x = TO_ECMA_NUMBER_T (exp (x));
x = TO_ECMA_NUMBER_T (floor (x));
x = TO_ECMA_NUMBER_T (log (x));
...
#else
x = TO_ECMA_NUMBER_T (absf (x));
x = TO_ECMA_NUMBER_T (acosf (x));
x = TO_ECMA_NUMBER_T (asinf (x));
x = TO_ECMA_NUMBER_T (atanf (x));
x = TO_ECMA_NUMBER_T (ceilf (x));
x = TO_ECMA_NUMBER_T (cosf (x));
x = TO_ECMA_NUMBER_T (expf (x));
x = TO_ECMA_NUMBER_T (floorf (x));
x = TO_ECMA_NUMBER_T (logf (x));
...
#endif

akosthekiss, 2018-04-27:
@DJLinux Thanks for looking up those math lib use cases. I've skimmed through the code base for `double`s, and actually found some more in the following functions:

API (@zherczeg suggested a redefinition-based workaround for these, as they cannot be changed):
- jerry_get_number_value
- jerry_create_number
- jerry_port_get_current_time

Internal functions using `double` instead of `ecma_number_t` (these may be worth changing, they may be actual errors):
- ecma_op_create_arraybuffer_object
- ecma_builtin_typedarray_prototype_set
- ecma_builtin_number_prototype_object_to_string

Floating point to string conversion (these are a hard nut to crack, I'm not sure we have a solution for "de-doubling" them right now as they are used for the stringification of whatever floating point format we use):
- ecma_number_to_decimal
- ecma_number_of_digits
- ecma_double_to_ascii
- ecma_double_to_binary_floating_point
- ecma_number_to_binary_floating_point_number
- ecma_normalize_high_prec_data
- ecma_multiply_high_prec_by_10
- ecma_divide_high_prec_by_10
- ecma_errol0_dtoa

PS: I don't think that our own math lib (jerry-libm) has float versions for the math functions. That may be a blocker for the rewrite that you suggest above. But the suggested redefinition trick may work for `fabs` -> `absf`, etc. as well.

PPS: as far as I can remember, CONFIG_ECMA_NUMBER_FLOAT32 and CONFIG_ECMA_NUMBER_FLOAT64 never promised not to use any other number representation during computations, only that ECMA numbers will be _stored_ on 32 or 64 bits when they have to be stored somewhere in memory. And that was also for memory and/or runtime efficiency reasons only. JerryScript was designed with full C99 compliance in mind, types included.

PPPS: All my above comments don't mean that we may not move towards purer 32/64 bit usage, though, if it can be nicely/efficiently/maintainably solved.

zherczeg, 2018-11-06:
I think this issue is resolved. Please reopen it if needed.

closed 2018-11-06 by zherczeg